intel-iot-devkit/tinyb #14: Dealing with incorrect data length in descriptor's value

IOOOTA, 2016-04-19:
Not really an issue, much more a support request.

Env:
- kernel 4.2.0 32bit (Ubuntu 15.10 Linux distro)
- bluez 5.39 (built from sources, with the --enable-experimental and --enable-library options)
- tinyb 0.4.1 (built from sources, with the -DCMAKE_BUILD_TYPE=DEBUG and -DBUILDJAVA=ON options)
- OpenJDK 1.8.66

Using a simple Java main program, I'm connecting to a GATT server peripheral (specifically a Xiaomi MiBand bracelet, latest version) exposing the Heart Rate Service (0x180D).
Inside the Hart Rate Measurement characteristic of the service (0x2A37) I'm trying to access the CCC descriptor (0x2902): I get the BluetoothGattDescriptor instance with no problem, but the following error occurs while trying to read the current value of the descriptor:

code is simply `de.getValue()` (where "de" is the BluetoothGattDescriptor object), and the outcome is

> (process:9020): GLib-CRITICAL **: g_bytes_get_data: assertion 'bytes != NULL' failed
> terminate called after throwing an instance of 'std::bad_alloc'
>   what():  std::bad_alloc

Using an external tool (nRF Master Control Panel on Android 5.0) I have found that the value field of that CCC descriptor has an "incorret length": nRF says "wrong size, 16 bits were expected (0x00)".
It seems that the manufacturer put a 8 bits value in the CCC descriptor instead of 16bits.

May that be the cause of the error reported above? I'm not a C++ developer, but it seems that "bad_alloc" deals with allocation, so maybe is a matter of size.

For completeness: the MiBand device also seems to have a problem with the pairing process, since bluez5 asks for the SC (Secure Connection) option and the bracelet fails the handshake with an "invalid parameter" response. Here is an excerpt of the btmon tool output:

> < ACL Data TX: Handle 64 flags 0x00 dlen 11
>       SMP: Pairing Request (0x01) len 6
>         IO capability: NoInputNoOutput (0x03)
>         OOB data: Authentication data not present (0x00)
>         Authentication requirement: Bonding, MITM, SC, No Keypresses (0x0d)
>         Max encryption key size: 16
>         Initiator key distribution: EncKey Sign LinkKey (0x0d)
>         Responder key distribution: EncKey IdKey Sign LinkKey (0x0f)
> > ACL Data RX: Handle 64 flags 0x02 dlen 6      SMP: Pairing Failed (0x05) len 1
>         Reason: Invalid parameters (0x0a)
> < HCI Command: Disconnect (0x01|0x0006) plen 3
>         Handle: 64
>         Reason: Authentication Failure (0x05)

I cannot say if this lack of support of the SC feature may be related to the read error or not...

I hope my questions can be of any help and not just a bothering.

Thanks for the very good code you are delivering to the community.

petreeftime, 2016-04-19:
Thanks for the report. It looks like an issue with the data received from BlueZ, so it's likely that it's the same issue that nRF reports, but I will try replicating to see if there's something I can fix on my side.

petreeftime, 2016-04-19:
Actually, I remembered seeing this error before. There is also the possibility that the device is not really connected or disconnects during operation and BlueZ returns an array of length zero. Running malloc(0) returns null, hence the failed assert.

Unfortunately, at this time it's not possible to get disconnection events, but I plan on adding this feature.

robertopierpaoli, 2016-04-19:
Well, it may be, but if you may see the test running you may find it unlikely, for two reasons:
- the whole mini-test executes in less than 5 seconds and the "suggested" reconnection time of the MiBand should be around 39s
- if I replace that specific instruction (`ch.getValue()`) with a different one (like `srv.getCharacteristics()`, where "srv" is the service) it works: in case of disconnection also `getCharacteristics` should fail, makes sense?

petreeftime, 2016-04-19:
Try using readValue instead. getValue returns a cached value, but since it might be the first read there would be no cached value. If this is the issue I'll catch the exception and return null instead of crash.

robertopierpaoli, 2016-04-19:
I've just tried `readValue()`, I don't get the value, this is the outcome:

> #
> # A fatal error has been detected by the Java Runtime Environment:
> #
> #  SIGSEGV (0xb) at pc=0x84cdfad0, pid=12137, tid=3058088768
> #
> # JRE version: OpenJDK Runtime Environment (8.0_66-b17) (build 1.8.0_66-internal-b17)
> # Java VM: OpenJDK Server VM (25.66-b17 mixed mode linux-x86 )
> # Problematic frame:
> # C  [libtinyb.so.0+0x5cad0]  gatt_descriptor1_call_read_value_sync+0xe1
> #
> # Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
> #
> # An error report file with more information is saved as:
> # /home/iooota/workspace/com.iooota.tests.tinyb/hs_err_pid12137.log
> #
> # If you would like to submit a bug report, please visit:
> #   http://bugreport.java.com/bugreport/crash.jsp
> # The crash happened outside the Java Virtual Machine in native code.
> # See problematic frame for where to report the bug.
> #

Here is the quoted log file.
[hs_err_pid12137.log.zip](https://github.com/intel-iot-devkit/tinyb/files/226160/hs_err_pid12137.log.zip)

petreeftime, 2016-04-19:
Ok. Thank you. I was able to replicate this and found some additional errors with getValue.

petreeftime, 2016-04-20:
It should be fixed in a99cba2755935b01c80cbd7b83eba3a6d788d5d6 and readValue should work as expected. Could you please confirm?

I will add an exception when a cached value for **getValue** is not available yet.

IOOOTA, 2016-04-20:
It seems fine, `readValue` causes no errors now, the returned value is just an empty array, so I assume that no value at all is present inside the descriptor: the smartphone app that I'm using as a comparison gives the following output

> Incorrect data length (16bit expected): (0x) 00

Maybe the value is considered "incorrect" since it is totally absent, while 16bits were expected.

Using `getValue` I get the same errore as before

> (process:25408): GLib-CRITICAL **: g_bytes_get_data: assertion 'bytes != NULL' failed
> terminate called after throwing an instance of 'std::bad_alloc'
>   what():  std::bad_alloc

but I understand that this is still expected at the moment, right?

petreeftime, 2016-04-20:
Great, I'll close the issue then.

closed 2016-04-20 by petreeftime