intel-iot-devkit/tinyb #15: TinyB not reporting GATT services when queried (HelloTinyB not working on Edison)

youcangetme, 2016-04-29:
Hello!

I have a project that needs to connect to an array of Bluetooth LE sensors. I wrote the sensors myself and they are compliant with the Environmental Sensing and Battery Service profiles. I can use my Android BLE apps to connect to the sensors and utilize them just great. In fact they work better than expected. Now I'm trying to get my app to run on the Edison with all the latest software update/upgraded with opkg.

I am trying to run a modified version of the HelloTinyB example. The modifications are to point it at the sensors I wrote. When I run it I get the following output;

Listening for transport dt_socket at address: 8005
The discovery started: true
Address = E2:E4:BD:03:41:03 Name = DSLT-TMP Connected = false
Address = F0:79:59:2B:93:40 Name = Snarf Connected = false
Error: GDBus.Error:org.bluez.Error.InProgress: Operation already in progress
Found device: Address = E2:E4:BD:03:41:03 Name = DSLT-TMP Connected = false
Sensor with the provided address connected
Services exposed by device:
Device Addr: E2:E4:BD:03:41:03
Device: DSLT-TMP -> 00001530-1212-efde-1523-785feabcd123
Device: DSLT-TMP -> 00001800-0000-1000-8000-00805f9b34fb
Device: DSLT-TMP -> 00001801-0000-1000-8000-00805f9b34fb
Device: DSLT-TMP -> 0000180a-0000-1000-8000-00805f9b34fb
Device: DSLT-TMP -> 0000180f-0000-1000-8000-00805f9b34fb
Device: DSLT-TMP -> 0000181a-0000-1000-8000-00805f9b34fb
Device: DSLT-TMP -> 6e400001-b5a3-f393-e0a9-e50e24dcca9e
Device: DSLT-TMP -> ee0c2080-8786-40ba-ab96-99b91ac981d8
Services > ? 0

So the Device: DSTL-TMP lines where added during debugging and show the UUIDs the devices is advertising BUT when the "bluetoothServices = device.getServices();" is executed it brings back a zero length list forever. Its in a loop with a 4 second sleep. I've tried different sleep intervals with no luck.

When you print the UUIDs you can clearly see the UUIDs for the two services I want to connect to;

UUID: Battery Service (0000180f-0000-1000-8000-00805f9b34fb)
UUID: Environmental Sensing  (0000181a-0000-1000-8000-00805f9b34fb)

In fact, the above UUIDs I got from the bluetoothctl tool so I'm at a loss now as the tool sees everything just fine. EVen if I revert all of my changes to the example it should at the very minimum is print a list of the services that are available and then complain it couldn't find what it wanted.

The line "Services > ? 0" is in the loop waiting on the services and the 0 is the length of the list for each loop.

petreeftime, 2016-04-29:
Hi!

What image are you using for Edison?
Can you verify if /etc/dbus-1/system.d/bluetooth.conf contains the following lines under  the ```<policy user="root">``` block?
```
    <allow send_interface="org.bluez.GattCharacteristic1"/>
    <allow send_interface="org.bluez.GattDescriptor1"/>
```

petreeftime, 2016-04-29:
Also, can you check that ```/lib/systemd/system/bluetooth.service``` contains:
```
ExecStart=/usr/lib/bluez5/bluetooth/bluetoothd -E
```
If the -E flag is not present please add it, then run:
```
systemctl daemon-reload
systemctl restart bluetooth.service
```

youcangetme, 2016-04-29:
OK, I was kind of in panic mode. Its rare once you reach such an impasse to recover from it since these technologies simply do not have the manpower behind them to address everyone's issues.

I resolved the issue (yay!) I had missed the part about starting the service in experimental mode. I have no idea why Github wasn't notifying me you guys answered my posts. Thank you very much for your prompt response. I came back in to remove this post BUT! Now I have more questions;

**No, my bluetooth.conf does NOT have those entries, should it?**

Again, thanks for the replies and being polite to me for not reading the docs ya'll wrote ;-)

petreeftime, 2016-04-29:
If you can't access Characteristics or Descriptors, then you should those lines and try again, after restarting the board. As far as I know, the latest software release for Edison, available at software.intel.com/iot has them enabled, although the -E flag is still not added as a default.

youcangetme, 2016-04-29:
Thanks! I am polling my sensors successfully. Still working through the notifications being pushed though but at least I'm able to read characteristics now.

closed 2016-04-29 by youcangetme