intel-iot-devkit/tinyb #16: support for BLE Server ?

kiranpradeep, 2016-04-30:
Thanks for the library. This is helpful..

Will this library support creating a BLE Server ? I couldn't find a class which helps to add characteristics and services.

petreeftime, 2016-05-02:
Hi,

At this time it is not possible to create a BLE server, but I will consider adding this in future versions of the library. If you would like to contribute this functionality it would be most helpful.

ptzafrir, 2017-08-21:
Are there any updates on this?

LucBerge, 2020-12-01:
I'm also really interested about running a BLE server in java

oset-bschuhma, 2022-10-25:
OK, so it's not **JUST** me!  I looked at this, wanting to create a Server and couldn't find any way to register services, descriptors, characteristics, etc.

Anyone else found a Java library that will allow me to create a GATT Server running on Raspberry Pi (i.e. NOT Android)?

oset-bschuhma, 2022-10-26:
@LucBerge - Thanks very much for the suggestion!!

LucBerge, 2022-11-07:
@oset-bschuhma I just found a maintained library working on Raspberry Pi: https://github.com/weliem/blessed-bluez