intel-iot-devkit/tinyb #17: tinyb.BluetoothDevice.getRssi() always returns "0"

IOOOTA, 2016-05-02:
_(Same environment as in #14, using Java bindings)_

In my execution environment the `getRssi()` method always returns a value of "0".

This is basically my little test:

```
this.adapter.startDiscovery();
device = this.adapter.find(null, address, Duration.ofSeconds(15));
if(device != null)
{
    log.debug("RSSI: {}", device.getRssi()); // Before discovery has stopped
}
this.adapter.stopDiscovery();
if(device != null)
{
    log.debug("RSSI: {}", device.getRssi()); // After discovery has stopped
}
```

Output is sistematically 

> "RSSI: 0"
> "RSSI: 0"

The following is what I get looking at the bluetoothctl interactive console:

> [CHG] Controller 5C:F3:70:6A:D1:B1 Discovering: yes
> [CHG] Device 4A:27:A8:D4:71:24 RSSI: -78
> [NEW] Device 78:7F:35:45:DA:72 78-7F-35-45-DA-72
> [CHG] Device 0C:F3:EE:00:97:75 RSSI: -74
> [CHG] Device A4:77:33:9C:9A:2B RSSI: -73
> [CHG] Device C8:0F:10:19:0D:52 RSSI: -61
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Key: 0x0157
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x00
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x89
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0xc4
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x4a
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0xbb
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x94
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x79
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x91
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x9e
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x1a
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0xcc
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x98
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x3f
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x15
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0xf4
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0xd5
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0xca
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x01
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0xc8
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x0f
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x10
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x19
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x0d
> [CHG] Device C8:0F:10:19:0D:52 ManufacturerData Value: 0x52
> [CHG] Device A0:BB:3E:A0:69:D5 RSSI: -68
> [CHG] Device A0:BB:3E:A0:69:D5 TxPower: -2
> [CHG] Device A0:14:3D:6B:F3:2E RSSI: -67
> [CHG] Device A0:14:3D:6B:F3:2E TxPower: 2
> [CHG] Device A0:14:3D:6B:F3:2E TxPower is nil
> [CHG] Device A0:14:3D:6B:F3:2E RSSI is nil
> [CHG] Device A0:BB:3E:A0:69:D5 TxPower is nil
> [CHG] Device A0:BB:3E:A0:69:D5 RSSI is nil
> [CHG] Device C8:0F:10:19:0D:52 RSSI is nil
> [CHG] Device A4:77:33:9C:9A:2B RSSI is nil
> [CHG] Device 0C:F3:EE:00:97:75 RSSI is nil
> [CHG] Device 78:7F:35:45:DA:72 RSSI is nil
> [CHG] Device 4A:27:A8:D4:71:24 RSSI is nil
> [CHG] Controller 5C:F3:70:6A:D1:B1 Discovering: no

I've noticed the "...RSSI is nil" outputs just before stopping the discovery: may it be related to the "0" output from the lib?

petreeftime, 2016-05-03:
RSSI is optional and is only available under certain conditions: http://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/device-api.txt. I will check to see if it's available while discovery is turned on, but from reading the code I think it should work as BlueZ advertises and I don't think I can otherwise fix it.

IOOOTA, 2016-05-03:
I think it would be fine if the getter method may return the current actual value during the discovery and zero otherwise. That would be perfectly coherent with the behavior of the "bluetoothctl" interactive shell provided by Bluez where you need to run the "scan" command if you want to take a look at the RSSI values.

petreeftime, 2016-05-03:
So, for me RSSI returns values different than zero while discovery is on. I added this line:
```
System.out.println(String.format("Device: %s RSSI=%d", device.getName(), device.getRssi()));
```
after this line: https://github.com/intel-iot-devkit/tinyb/blob/master/examples/java/HelloTinyB.java#L38
and I get the output:
```
Device: CC2650 SensorTag RSSI=-59
```

closed 2016-05-03 by petreeftime

petreeftime, 2016-05-05:
If you wait a bit (5-10 seconds) before getting the RSSI value, is it different than zero afterwards?

IOOOTA, 2016-05-05:
I was trying right now, in my system I'm getting two different casistics : 

_(Note: in both cases, discovery is ON)_
- the device is already "known" by the adapter => the RSSI gets available after 2 seconds of discovery
- the device is "new" to the adapter => it takes a little less than 2 secondo to get the device, but then the RSSI is immediately available, I don't need to wait.

In both cases, if the discovery if OFF the RSSI is 0.

So, in general, in my system it seems that more or less 2 seconds of the discovery process are needed to have the RSSI available: I'm adjusting my code to this timings.

IOOOTA, 2016-05-05:
Ok, I made a step forward: now I can get the RSSI, in the worst case I have to perform a 2 seconds discovery before invoking the getter method.

Now the problem is with the RSSI value update: the value returned changes **very slowly**, like every 10-15 seconds. So, if I monitor the value for 30 seconds, the value gets updated only 2-3 times.
It's not something related to the library: the bluetoothctl console shows the same change frequency.

I have tried to force the update to be more frequent restarting the discovery, with the following code:

```
1   System.out.println("Handling discovery");
2   if(adapt.getDiscovering())
3   {
4 	   System.out.println("Stopping discovery");
5	   if(adapt.stopDiscovery())
6	   {
7		   TimeUnit.MILLISECONDS.sleep(50);
8		   System.out.println("Restarting discovery");
9		   adapt.startDiscovery();
10	   }
11  }
12  else
13  {
14	   System.out.println("Starting discovery");
15	   adapt.startDiscovery();
16  }
      // here I read the RSSI
```

This way I have observed some strange behaviors:

- some times `getDiscovering` returns "false", **but the discovery is actually running**, infact `startDiscovery` is called (line 15) and I get a 

> Error: GDBus.Error:org.bluez.Error.InProgress: Operation already in progress

- after stopping the discovery, I need to wait some microseconds or the process will stall, without any error in the console: this is the reason of the `sleep` instruction (line 7)

With this (quite ugly) trick I get an updated value every each 2 seconds, more or less.

Anyway I don't think this is the right way to go: do you think you may refactor the getRssi() function so that it reads the current value (if available) in a synchronous fashion?

IOOOTA, 2016-05-17:
Hi @petreeftime, I know this issue is marked as "closed", but I would be happy to have your feedback about my last comments, if possibile.

Thank you very much in advance.

petreeftime, 2016-05-17:
Actually, I'm working on getting notifications working. It might be possible have notifications for when a property changes, which might be more useful than waiting for RSSI to get updated. It might take a while to make things work properly.

IOOOTA, 2016-05-17:
That's great, I was just curious about what's going on. 

FYI, I'm writing with another team that is using Bluez over DBus (though not using TinyB) and they complain that the "StartNofity" method enables both notifications and indications, without letting the developer distinguish the two casistics: maybe are you experiencing the same behavior? 

Thanks for the great work.

petreeftime, 2016-05-17:
I have not tried, but the BlueZ documentation suggests that it's the expected behavior: http://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/gatt-api.txt#n88. I think BlueZ handles the ACK for indications as well, so they should be the same as far as the applications is concerned.