intel-iot-devkit/tinyb #18: getModalias and getIcon on BluetoothDevice cause a failure

IOOOTA, 2016-05-02:
_(Same environment as in  #14, using Java bindings)_

Invoking getModalias or getIcon on a device causes an error that is not propagated as a Java exception.

The following is the output I get in the debug console:

> terminate called after throwing an instance of 'std::logic_error'
>   what():  basic_string::_M_construct null not valid

Same output for the two methods.

Note: the device is not connected when invoking those methods, but I guess that should not make any difference.

petreeftime, 2016-05-03:
Icon and modAlias (and a few other properties) are optional, but are treated as always available. I will make it such that they return null when they are not available.

petreeftime, 2016-05-12:
Fixed in https://github.com/intel-iot-devkit/tinyb/commit/b0545a46bcfec1571c75499f0f2ed319f1f73c2e.

closed 2016-05-12 by petreeftime