intel-iot-devkit/tinyb #21: Building from Git for Java and getting errors...

youcangetme, 2016-05-03:
I'm not exactly the brightest bulb in the room so trying to keep up with the newest code I tried to compile on my Edison and got the following errors. Could you expand on the build instructions so us moron Java guys don't poke an eye out or blow something up?

Thanks

```
root@edison:~/tinyb/build# cmake -DBUILDJAVA=ON ..

CMake Error at /usr/share/cmake-3.3/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Java (missing: Java_JAR_EXECUTABLE Java_JAVAH_EXECUTABLE
  Java_JAVADOC_EXECUTABLE) (found version "1.8.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.3/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.3/Modules/FindJava.cmake:209 (find_package_handle_standard_args)
  java/CMakeLists.txt:1 (find_package)

```

madhead, 2016-05-03:
Are you using JRE or JDK? Do you have `JAVA_HOME` environment variable pointing to the **JDK**?

youcangetme, 2016-05-03:
root@edison:~/tinyb/build# which java
/usr/java/jdk1.8.0_91//bin/java
root@edison:~/tinyb/build# which javac
/usr/java/jdk1.8.0_91//bin/javac

It's the Oracle version instead of that terrible OpenJVm stuff, is that a problem?

youcangetme, 2016-05-04:
OK, figured it out. Thanks Anyways!

closed 2016-05-04 by youcangetme