intel-iot-devkit/tinyb #24: Building on a Raspberry PI3 (raspbian 8 jessie)

IOOOTAlan, 2016-05-06:
Greetings,
I just tried to build this project on a Raspberry PI 3 and found the following problems with the packages/tools available on the Raspbian 8 distribution:

- the cmake version shipped with raspbian is 3.0.2, while this project requires at least versione 3.1.0 to build
- the gcc version present is v4.9.2, which needs a `-std=c++11` flag to correctly build the C++ code in this project

After solving these two issues, the build went on smoothly.
While the second issue is avoidable by a simple override of the CMAKE_CXX_FLAGS variable during the configuration phase, the first issue requires a change in the main CMakeLists.txt file.
I would like to ask if there is a reason for setting the minimal cmake version to v.3.1.0.
If not, I would gladly submit a pull request to make the build possible on cmake versions lower than 3.1.0.

petreeftime, 2016-05-06:
I wanted the project to build with multiple compilers, not only GCC. CMake provides a way to do this with the [CXX_STANDARD](https://github.com/intel-iot-devkit/tinyb/blob/master/src/CMakeLists.txt#L46) keyword, but this is only available in CMake 3.1+.

That aside, if the project builds correctly on Travis with clang and gcc and generates no additional warnings or errors, I will merge your patch, since it would make it easier to build on older distros.

IOOOTAlan, 2016-05-09:
Ok, then there's an incoming pull request ;-)

petreeftime, 2016-05-09:
Fixed with #25 .

closed 2016-05-09 by petreeftime