intel-iot-devkit/tinyb #26: JVM crash on std::runtime_error exceptions

IOOOTAlan, 2016-05-09:
Greetings,
we're currently facing a number of JVM crashes while using tinyb in our OSGI development environment. The crashes seem to be all caused by exceptions thrown inside the native C++ portion of tinyb.
Upon further investigation (see [this StackOverflow post](http://stackoverflow.com/questions/4138168/what-happens-when-i-throw-a-c-exception-from-a-native-java-method), the first answers seem pretty informative), the root cause for these crashes seems to be that the JNI specification is a C-only specification which is totally agnostic to C++ constructs (like exceptions): the behavior in these cases is undefined in the spec and could also bring down the JVM (like what's happening to us).
The ideal way to deal with native exception inside a JNI application would be to catch all the incoming C++ exceptions inside the native code and then translate those native exception/errors in proper return codes or Java exceptions/errors when needed.
From what I've seen in tinyb source files, there are a number of exceptions like this:

```c++
if (!result)
{
    throw std::runtime_error("cannot create instance of class\n");
}
```

that get directly thrown to the JNI runtime. These would all need to be caught (or, in this case, directly translated to Java exceptions) for the crashes to disappear.

petreeftime, 2016-05-09:
I'm currently working on getting C++ exceptions converted to Java, but it requires adding exception handling code to each JNI interface function, so it might take a few days to complete.

petreeftime, 2016-05-10:
Should be fixed in 44ea9ff1da0e10a7e0e8c14167f0fc7e8937bd3a. It's a huge patch, if you have any issues please tell me. I will also try to throw errors from BlueZ/DBus as IOExceptions and mark Java functions with the likely and useful exceptions.

IOOOTAlan, 2016-05-10:
That's great!
We're going to test it right now

IOOOTAlan, 2016-05-27:
We're currently facing an issue on some of our environments where the same runtime exception is always thrown.
I'm going to open a new issue about it, because I'm not entirely sure it's related to the changes introduced in 44ea9ff.

IOOOTAlan, 2016-06-01:
I guess we can close this issue along #30 too.

closed 2016-06-01 by petreeftime