intel-iot-devkit/tinyb #27: Trouble reading services and characteristics

youcangetme, 2016-05-17:
I had a modified HelloTinyB example that reads an Environmental Sensing service and the temperature characteristic. After a week or so I cam back to the project and now it does not run, no changes. It connects fine to the device but when it starts to read the services it gets back a empty list of services and spins until shutdown.

I keep the Edson up to date using opkg update + opkg upgrade.

I tried to compile the TinyB project on one Edison but for whatever reason the Intel version of Eclipse would not find the native libs when it executed the my project remotely so I gave up on it. I'm not a sys admin guy, I code for a living so forgive me.

The odd part is in I go into the bluetoothctl and do a info on the device I see the services. I mean it pops up like lightening fast, no delay.

Is there a how-to doc on using the compiled version of TinyB on the Edison with Java so I can use the newest version? I'm still very fuzzy on how the Intel Eclipse connects these external resources at runtime. I try to read all the crazy stuff in the scripts but I would like to know how ya'll do it. In fact I would like to dump the Intel eclipse stuff entirely and use MyEclipse + Spring Boot w/Maven for this project.

petreeftime, 2016-05-17:
TinyB does not have a Maven repo yet, However, if you are using Java, you just need to copy tinyb.jar to your Eclipse project. At runtime you will need to have TinyB in your classpath (on Edison it should be /usr/lib/java/tinyb.jar). Otherwise, it should work as intended.

As far as I know you should be able to compile it on the board using the guide in the [README](README.md). Otherwise you will need to compile it with the Edison toolchain which you can find here: https://software.intel.com/en-us/iot/hardware/edison/downloads, under SDK. (cmake, then make and make install)

Otherwise, I don't exactly now what the issue could be. The steps in #15 should pretty much cover the know issues that I know of. It's possible that opkg upgrade overwrote some file.

youcangetme, 2016-05-17:
*facepalm* OK, my bad. I swear I'm not normally this dumb ;-) Thanks for being kind to me.

youcangetme, 2016-05-17:
I compiled the project again and enabled the experimental -E flag and when I run from the Intel IoT Eclipse after replacing the old tinyb.jar with the newly compiled one I get the following;

```
Exception in thread "main" java.lang.UnsatisfiedLinkError: tinyb.BluetoothManager.getNativeAPIVersion()Ljava/lang/String;
	at tinyb.BluetoothManager.getNativeAPIVersion(Native Method)
	at tinyb.BluetoothManager.getBluetoothManager(BluetoothManager.java:219)
	at HelloTinyB.main(HelloTinyB.java:127)
```

I just copied the jar file, should I have symlinked it instead?

petreeftime, 2016-05-18:
You also need to copy the libtinyb.so* and libjavatinyb.so* files to the /usr/lib/ folder on the Edison.

youcangetme, 2016-05-18:
I compiled the project on the Edison. I followed the recipe all the way through to the `make install` with no errors. The libs are in place where you describe;

```
lrwxrwxrwx 1 root root 13 Mar 14 22:36 /usr/lib/libtinyb.so -> libtinyb.so.0
lrwxrwxrwx 1 root root 17 Mar 14 22:36 /usr/lib/libjavatinyb.so -> libjavatinyb.so.0
```

Is this a problem with the Intel Eclipse deploying the and running the app then?

--jb

petreeftime, 2016-05-18:
What version of Intel Eclipse are you using? What command is using to launch the program? It should leave a file in /tmp/ which gives you the run command. If the command specifies a -Djava.library.path option you should try upgrading to a newer version.

youcangetme, 2016-05-18:
Oh boy. I went into the Intel page for the downloads to get a newer version of the Eclipse IDE and links are all 404.

This is the last straw on this IDE thing. I will move the project over to MyEcllipse IDE which makes uploading and execution slow and manual but, oh well! At least I can figure out for myself what is going wrong.

--jb

petreeftime, 2016-05-18:
Could you tell me what links you are having issues with? Thanks.

youcangetme, 2016-05-18:
https://software.intel.com/en-us/iot/software/ide

Scroll down to the "Intel® System Studio IoT Edition" section and Windows, Mac and Linux links all 404

--jb

youcangetme, 2016-05-18:
I am now starting my Spring Boot based app using the following;

java -Dloader.path=lib/,config/,/usr/lib/java/ -jar TerraStill-1.0.0-SNAPSHOT.jar

I still get the following exception when I start looking fro BLE devices;

Caused by: java.lang.UnsatisfiedLinkError: tinyb.BluetoothManager.getNativeAPIVersion()Ljava/lang/String;
        at tinyb.BluetoothManager.getNativeAPIVersion(Native Method)
        at tinyb.BluetoothManager.getBluetoothManager(BluetoothManager.java:219)
        at com.subnoize.terrastill.HelloTinyB.start(HelloTinyB.java:158)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)

Maybe something has gone terribly wrong with my make install? I see the *.so are there. They are symlinked to another symlink to the lib that has the version number (I assume that the version number) OR is that the problem?

youcangetme, 2016-05-18:
When I do a build on my Edison I see the install going everywhere BUT the /usr/lib directory

```
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/DESTINATION/tinyb.jar
-- Installing: /usr/local/lib/libjavatinyb.so.0.4.2
-- Up-to-date: /usr/local/lib/libjavatinyb.so.0
-- Up-to-date: /usr/local/lib/libjavatinyb.so
-- Removed runtime path from "/usr/local/lib/libjavatinyb.so.0.4.2"
-- Up-to-date: /usr/local/include/tinyb.hpp
-- Up-to-date: /usr/local/include/tinyb
-- Up-to-date: /usr/local/include/tinyb/BluetoothGattDescriptor.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothGattService.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothManager.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothAdapter.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothDevice.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothGattCharacteristic.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothEvent.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothObject.hpp
-- Installing: /usr/local/lib/pkgconfig/tinyb.pc
-- Installing: /usr/local/lib/libtinyb.so.0.4.2
-- Up-to-date: /usr/local/lib/libtinyb.so.0
-- Up-to-date: /usr/local/lib/libtinyb.so

```

petreeftime, 2016-05-18:
I think I might have to update the README, can you try ``` -DCMAKE_INSTALL_PREFIX=/usr ```? (without :PATH).

youcangetme, 2016-05-18:
I made the changes to the cmake line and recompiled and installed;

youcangetme, 2016-05-18:
I made the changes and recompiled and installed;

```
2016-05-18 21:23:35.581 DEBUG 2038 --- [           main] com.subnoize.terrastill.HelloTinyB       : >>>>>>> Starting HelloTinyB
Native code library failed to load.
java.lang.UnsatisfiedLinkError: /usr/lib/libjavatinyb.so.0.4.2: libawt.so: cannot open shared object file: No such file or directory
```

no luck :-(

petreeftime, 2016-05-19:
Hi,

I tried to compile and install directly on Edison, rather than through the toolchain as usual, and I found out that the jar was installed in the wrong location. I reverted to the old way of installing the jar, and now it works properly.
This series of steps works for me on the Edison 3.0 image, executed in the directory where I downloaded the tinyb repo:
```
export PATH=/usr/lib/jvm/java-8-openjdk/bin/:$PATH
export JAVA_HOME=/usr/lib/jvm/java-8-openjdk/
mkdir build
cd build
cmake -DBUILDJAVA=ON -DCMAKE_INSTALL_PREFIX=/usr ..
make install
```
Please tell me if this works for you.

Thanks for the report on the wrong links to the Eclipse IDE. I have submitted a ticket to the team which should be able to fix this. I'll update the docs later today, to include the issues you helped me discover.

youcangetme, 2016-05-20:
Thank you so much for helping. **Here is what I just learned!**

I compiled first using the Oracle JDK 1.8. TinyB compiled with no errors. I ran my code and it threw the same errors. Still broken.

Then I went back and compiled exactly as you did **BUT** this time I used the OpenJDK. It compiled and when I ran it, it worked just fine.

That really got me thinking so I rebooted to reset all of the paths back as I had them before and I ran the app again using the Oracle JDK and the native code libraries weren't found!

This means there is a difference between the Oracle and OpenJDK flavors. Now I prefer the Oracle for its better tuned Hotspot but I'm not above moving over to the OpenJDK until we can figure out the differences.

Anyways, I have a Spring Boot v1.3.1 app running using Bluetooth on an Intel Edison and I'm as happy as a clam!

Thank you for helping. I would certainly note the differences between JVMs in your docs so other Java guys and girls won't think TinyB is broken.

Thank you for supporting Java and helping me out.

--jb

petreeftime, 2016-05-20:
Hotspot only runs on the target platform, so if you run on Edison with OpenJDK it does not really matter. As far as I know the native code should be compatible, and I don't know why libawt is linked by Oracle's JDK, but I will look into it.

I'll close this issue now, if you have any more questions please open another issue or reopen this one.

closed 2016-05-20 by petreeftime