intel-iot-devkit/tinyb #29: Support to asynchronous notification of events?

dbonino, 2016-05-23:
Is it possible to add, in Java bindings, support to notification events generated by the Bluez stack when a device notifies a new characteristic value? E.g, by offering an interface to register a listener to Bluetooth Events generated by the Bluez stack when devices are notifying...

petreeftime, 2016-05-23:
It's a feature I want to include in the next version, but have not yet started work on it.

dbonino, 2016-05-23:
That will be really useful, thanks

youcangetme, 2016-06-06:
I just got to this point in my code and spent the morning tracking BluetoothCallback and BluetoothEvent objects thinking those would lead me somewhere. The Bluetooth hardware I developed is sending notifications and it would be much easier to let the Java side respond to those notifications as oppose to reading in a loop.

So development doesn't grind to a halt I will do the loop and read thing but count me in as needing this feature ASAP.

petreeftime, 2016-06-13:
This is my top priority right now, and I'm hoping to have a working version for Java soon, as the C++ version is mostly done already.

petreeftime, 2016-07-12:
Resolved in  1f5082f

closed 2016-07-12 by petreeftime