intel-iot-devkit/tinyb #30: "Trying to acquire null object" std::runtime_error on every run

IOOOTAlan, 2016-05-27:
Greetings,
following the comment on #26, I'm opening a new issue on this.
We're currently testing the latest development builds of tinyb on different systems (x86, ARM) and we're constantly facing the aforementioned error only on our ARM boards.
That error is thrown as soon as some operation on the adapter object obtained from the BluetoothManager is invoked from our Java code. We performed various debugging sessions and we've seen it consistently happening on the first call to some modified of the bluetooth adapter instance, like `BluetoothAdapter.getPowered()`, but also `BluetoothManager.setDefaultAdapter()`.
Using gdb I was able to determine that the problem seems to arise from the `getInstanceField` function inside the [helper.cxx](https://github.com/intel-iot-devkit/tinyb/blob/863cf2d30edd15d6909477d2d2604b65fa68307a/java/jni/helper.cxx) file: this functions returns a null pointer when called inside the other functions/methods, like `getInstance`, `setInstance` etc., causing the runtime_error to be thrown.
Here are some additional notes:
 - I tried using different bluez versions (5.23, 5.36, 5.37, 5.40) to no avail;
 - the bluetooth adapter is indeed working using bluetoothctl from the command line;
 - the bluetooth adapter is correctly found by tinyb, but any other operation on it raises the error;
 - [correct DBUS policies](https://github.com/intel-iot-devkit/tinyb/blob/2f8c5c7f0f26cd2a1144000ba1c0f6ea501cc79e/TROUBLESHOOTING.md) for Bluez GATT interfaces are in place;
 - this happens on kernel v4.4 and v4.5

Given that we've seen this issue happening only on ARM hardware, could it be something endianness-related?

petreeftime, 2016-05-27:
Hi Alan, are you running on 32bit ARM or 64bit, and what Java version are you running and compiling with?

We'll look into the issue, it seems really weird :)

IOOOTAlan, 2016-05-27:
Sorry, forgot to mention that kind of info.
We're running on ARMv7 processor (32 bit) and using Oracle JDK 8u91.

petreeftime, 2016-05-27:
This might be a long shot, but can you try this patch: https://github.com/intel-iot-devkit/tinyb/commit/ce6896d91b526d660bc599fa80e05d9adf308194?

IOOOTAlan, 2016-05-27:
I did a couple of tests but the error stays the same.

IOOOTAlan, 2016-05-30:
I also tried using OpenJDK 8u91 (to exclude issues with the JNI implementation in Oracle JVM), but I get the same error.

petreeftime, 2016-05-30:
Did previous versions work, or was the bug introduced recently?

IOOOTAlan, 2016-05-30:
I don't know, I only did tests on ARM hardware with the latest versions.
Will try to see if I can arrange some tests on older versions...

petreeftime, 2016-05-31:
My first commit fixed one of the issues, but it seems ```long``` has been used instead of ```jlong``` in a few other places. It seems to be the root cause of the issue and I should hope to have a fix out soon.

petreeftime, 2016-05-31:
I have pushed the commit with the identified issues:  a739bf4, hopefully it fixes your problems. Let me know if the issues persist.

IOOOTAlan, 2016-05-31:
It's definitely a step in the right direction, because now I'm able to scan for devices and perform other operations that before caused the runtime error.
However, on the first call to `BluetoothGattCharacteristic.readValue` a SIGSEGV is raised and the entire JVM is torn down. The stacktrace is this:
```
#0  __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
#1  0x76ebb638 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#2  0x76ebc33a in __GI_abort () at abort.c:89
#3  0x767f24fc in os::abort(bool) () from /opt/jdk1.8.0_91/jre/lib/arm/server/libjvm.so
#4  0x7695b594 in VMError::report_and_die() () from /opt/jdk1.8.0_91/jre/lib/arm/server/libjvm.so
#5  0x767fc494 in JVM_handle_linux_signal () from /opt/jdk1.8.0_91/jre/lib/arm/server/libjvm.so
#6  0x767ef9d8 in signalHandler(int, siginfo*, void*) () from /opt/jdk1.8.0_91/jre/lib/arm/server/libjvm.so
#7  <signal handler called>
#8  0x76620d40 in jni_ThrowNew () from /opt/jdk1.8.0_91/jre/lib/arm/server/libjvm.so
#9  0x501cdfba in JNIEnv_::ThrowNew (this=0xc10d3c, clazz=0x0, 
    msg=0x10e0dd4 "GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method \"ReadValue\" with signature \"\" on interface \"org.bluez.GattCharacteristic1\" doesn't exist\n") at /opt/jdk1.8.0_91/include/jni.h:822
#10 0x501cde8c in raise_java_bluetooth_exception (env=0xc10d3c, e=...) at /home/ubuntu/sorgenti/tinyb/java/jni/helper.cxx:214
#11 0x501ce3fe in Java_tinyb_BluetoothGattCharacteristic_readValue (env=0xc10d3c, obj=0x4b5038ec) at /home/ubuntu/sorgenti/tinyb/java/jni/BluetoothGattCharacteristic.cxx:89
#12 0x73dcb9d0 in ?? ()
```
It seems like I am somehow missing the ReadValue method with an empty signature in the Bluez DBUS interface...

petreeftime, 2016-05-31:
That seems to be a compound issue. First, it can't find ```ReadValue```, which is very weird, but then, it can't find the ```tinyb/BluetoothException``` class to throw an exception. I've just tried making readValue throw an exception from C++, but it seems fine, so I'll have to look into it. But it seems to be a separate issue

petreeftime, 2016-05-31:
Apparently BlueZ recently updated the ReadValue method to take arguments. I'll have to figure out how to make it work in a compatible mode.

IOOOTAlan, 2016-05-31:
I got it to work (more or less) by downgrading my bluez version from 5.40 to 5.37 (default version bundled with Ubuntu 16.04). I also noticed that I was using an outdated tinyb.jar version, so these two actions brought me to an almost working state.
There are still some issues with specific GATT characteristics, but nothing that causes a JVM crash for now.
I'm now compiling version 5.39 to further test those GATT characteristics.

petreeftime, 2016-06-01:
I'll close this issue and start a new one for 5.40 compatibility.

closed 2016-06-01 by petreeftime