intel-iot-devkit/tinyb #35: Java BluetoothException handling

IOOOTAlan, 2016-06-01:
During some tests with a BLE device, we found that some commands were not being sent to it.
Upon further inspection, we found that the reason for this was that the `BluetoothDevice.connect()` method was "silently" (meaning that it was not clearly visible) throwing a BluetoothException for an underlying "Timeout while connecting" error.
Given that the `tinyb.BluetoothException` class is package private, we were only able to catch that exception using a catch all statement on the basic Exception class.
I would like like to know if there is a reason for the current visibility setting for the `BluetoothException` class and also ask if the signature of the various methods that can indeed throw such an exception shall be modified as well to reflect it.

petreeftime, 2016-06-01:
Actually, I wanted to mark the functions which might throw this sort of exception with throws first, but I guess I'll do this later. Changed in patch 5b1524bc06a3d4cfea0dab9690880040499813c. I'll keep this issue open for now.

petreeftime, 2016-08-22:
This has been fixed for a while.

closed 2016-08-22 by petreeftime