intel-iot-devkit/tinyb #36: BluetoothManager::get_devices() returns devices that are no longer present

whbruce, 2016-06-22:
BluetoothManager::get_devices() seems to return a list of all devices it ever seen rather than the ones that are currently in range. 

Expected behavior should be similar to `hcitool lescan` which just lists devices in range.

petreeftime, 2016-06-23:
```BluetoothManager::get_devices()``` does not start a scan, it returns the list of devices BlueZ knows about, whether they are recently scanned or have been connected to in the past. This is the expected behavior as per BlueZ or blueman-manager. A function like ```get_nearby_devices``` which starts a scan and checks that the devices are also in range might clear this up.

whbruce, 2016-06-23:
I call `BluetoothManager::start_discovery()` and wait several seconds before calling `BluetoothManager::get_devices()` but still get previously connected devices that are no longer present. What is the best way to filter them out? I can try to connect and wait for failure but that takes too long.

petreeftime, 2016-06-24:
Yes, after you start discovery, they should get an RSSI value different than 0 after a while. I'm not sure there's any other way to do this right now.

whbruce, 2016-06-25:
Thanks for RSSI tip. I found out that you have to call `BluetoothDevice::get_rssi()` before `BluetoothManager::stop_discovery()` to get valid values. 

A `get_nearby_devices()` call would do this for you, so tag this issue as a feature request.

lsimedia, 2017-08-12:
I implemented an additional method called removeDevices() in the adaptor class, which will remove all known devices from internal list (so getDevice() will return an empty list). I call this method just before calling startDiscovery() so I will receive only the current nearby beacons.
I've implemented it on the java side only (for c++ the method remove_device was already present)...

framspott94, 2017-08-14:
Hi Isimedia,

is the feature you implemented, already in the library? I eventually need this feature very much, therefore it would be great, if you could explain to me how I can use this function.

All the best

framspott94

lsimedia, 2017-08-14:
I've done a pull request with the modification, so the maintainer can review the code and merge it if needed. For the moment you can clone my tinyb fork.
The method to call is
int cnt = manager.getAdapters().get(0).removeDevices();

framspott94, 2017-08-14:
Thanks for the answer! I cloned your fork and used it succesfully!

All the best - and thanks for the feature!

framspott94

sbodmer, 2017-08-17:
I have added a new method called startNearbyDiscovery() in the main BluetoothManager class with a listener as argument to be notified when a new device is found until stopNearbyDiscovery() is called.
This approach is useful in my use case (and perhaps your's too...)