intel-iot-devkit/tinyb #37: Make fails on Raspbian Jesse 8

bokas74, 2016-06-26:
Hi, I'm new to this, so maybe I'm missing some libraries or configurations. Cmake ran fine but I'm getting errors during make. Here's the console output for cmake:

pi@raspberrypi:~/Downloads/tinyb-notifications/build $ cmake -DCMAKE_C_COMPILER=/usr/bin/clang -DCMAKE_CXX_COMPILER=/usr/bin/clang++ DBUILDJAVA=ON -DCMAKE_BUILD_TYPE=DEBUG ..
-- The C compiler identification is Clang 3.5.0
-- The CXX compiler identification is Clang 3.5.0
-- Check for working C compiler: /usr/bin/clang
-- Check for working C compiler: /usr/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'glib-2.0>=2.40'
--   found glib-2.0, version 2.42.1
-- checking for module 'gio-2.0>=2.40'
--   found gio-2.0, version 2.42.1
-- checking for module 'gio-unix-2.0>=2.40'
--   found gio-unix-2.0, version 2.42.1
CMake Warning at CMakeLists.txt:26 (message):
   - Install git to compile a production libtinyb!

INFO - libtinyb Version v0.4.3-dirty
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.8") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Downloads/tinyb-notifications/build

_However during make, I get:_

pi@raspberrypi:~/Downloads/tinyb-notifications/build $ make
Scanning dependencies of target tinyb
[  6%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothObject.cpp.o
In file included from /home/pi/Downloads/tinyb-notifications/src/BluetoothObject.cpp:25:
In file included from /home/pi/Downloads/tinyb-notifications/api/tinyb/BluetoothObject.hpp:27:
In file included from /usr/bin/../lib/gcc/arm-linux-gnueabihf/4.9/../../../../include/c++/4.9/mutex:35:
/usr/bin/../lib/gcc/arm-linux-gnueabihf/4.9/../../../../include/c++/4.9/bits/c++0x_warning.h:32:2: error: This file requires compiler and library support for the ISO C++ 2011 standard. This support
      is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
#error This file requires compiler and library support for the \
 ^
In file included from /home/pi/Downloads/tinyb-notifications/src/BluetoothObject.cpp:25:
In file included from /home/pi/Downloads/tinyb-notifications/api/tinyb/BluetoothObject.hpp:28:
In file included from /usr/bin/../lib/gcc/arm-linux-gnueabihf/4.9/../../../../include/c++/4.9/atomic:41:
/usr/bin/../lib/gcc/arm-linux-gnueabihf/4.9/../../../../include/c++/4.9/bits/atomic_base.h:74:3: error: unknown type name 'constexpr'
  constexpr memory_order
  ^
/usr/bin/../lib/gcc/arm-linux-gnueabihf/4.9/../../../../include/c++/4.9/bits/atomic_base.h:74:25: error: expected ';' after top level declarator
  constexpr memory_order
                        ^
/usr/bin/../lib/gcc/arm-linux-gnueabihf/4.9/../../../../include/c++/4.9/bits/atomic_base.h:124:11: error: unknown type name '__atomic_base'
  typedef __atomic_base<char>                   atomic_char;

_Any help/guidance is very welcome.

Thanks!_

doitdagi, 2016-06-26:
HI bokas74

I also have been trying to compile Tinyb for Raspbian this week, and faced the same problem. I am not sure if it is the efficient solution but I included the following compile option on the CmakeList.txt file and manged to compile. 
               add_compile_options(-std=c++11)

Hope it helps.

bokas74, 2016-06-26:
Thanks! Yeah was also trying other option, in the end I installed gcc+-6
(https://solarianprogrammer.com/2016/06/24/raspberry-pi-raspbian-install-gcc-compile-cpp-14-and-cpp-17-programs/).

And ended up compiling as 
cmake -DCMAKE_CXX_COMPILER=/usr/bin/g++-6 DBUILDJAVA=ON -DCMAKE_BUILD_TYPE=DEBUG ..
 
After that make, make install went thru, but with some code level warnings (unused variables mainly).

Thx again!

closed 2016-06-26 by bokas74

IOOOTAlan, 2016-06-27:
Hello,
you could have also used the `CMAKE_CXX_FLAGS` variable to only add the "-std=c++11" options for the gcc compiler.
See issue #24