intel-iot-devkit/tinyb #39: Bluetooth manager find does not time out

rlaybourn, 2016-06-29:
I am finding that in the example program asynctinyb.cpp the bluetooth manager find function does not seem to timeout if the device is not found . The program is stuck until I get the following error (I added some output to track progress)

startup
looking for device
terminate called after throwing an instance of 'tinyb::BluetoothException'
  what():  GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
found deviceAborted

the relevent section of code is bellow

std::cout << "startup\n";  
 BluetoothManager *manager = nullptr;                                                //create manager
  try {
        manager = BluetoothManager::get_bluetooth_manager();
    } catch(const std::runtime_error& e) {
        std::cout << "Error while initializing libtinyb: " << e.what() << std::endl;
        exit(1);
    }

    bool ret = manager->start_discovery();
    std::unique_ptr<BluetoothGattService> temperature_service;   //create service

    std::string device_mac(argv[1]);   
    std::cout << "looking for device\n";                                                       
    auto sensor_tag = manager->find<BluetoothDevice>( nullptr, &device_mac,  nullptr,  std::chrono::milliseconds(3000));
    std::cout << "found device";
    if (sensor_tag ==  nullptr) {
        std::cout << "Device not found" << std::endl;
        return 1;
    }

petreeftime, 2016-06-29:
Hi, timeout works ok as far as I tested some time ago, but it's possible that it gets stuck in some BlueZ function, since the issue is that it returns a org.bluez.Error.Failed.

Can you tell me what version of BlueZ are you using, and does the error remain after restarting the BlueZ daemon?

rlaybourn, 2016-06-30:
I am using blueZ version 5.39. Restarting the daemon does not have any affect on the problem. I did downgrade from bluez 5.4 so I guess it is possible there is something of that still laying around however everything else seems to be working fine. For now I am working around using a SIGALARM to implement a timeout.