intel-iot-devkit/tinyb #40: Java runtime not resolving all native components

dfclark, 2016-08-08:
I've built the tinyb library and am trying to build and run with the HelloTinyB example program, using Netbeans 8.  I'm hoping you can offer some insight, as I'm getting this error from the Java runtime:
"Native code library failed to load.
 java.lang.UnsatisfiedLinkError: /usr/local/lib/libjavatinyb.so.0.5.0: libtinyb.so.0: cannot open shared  object file: No such file or directory"

The java.library.path is defined for this project in Netbeans as:
    -Djava.library.path=/usr/local/lib/:/usr/local/lib/java/

The native files exist in /usr/local/lib, as shown by this output from the find command:
pi@raspberrypi:~/NetBeansProjects $ sudo find / -name *tinyb.so*
/usr/local/lib/libjavatinyb.so.0.5.0
/usr/local/lib/libjavatinyb.so.0
/usr/local/lib/libtinyb.so
/usr/local/lib/libtinyb.so.0
/usr/local/lib/libtinyb.so.0.5.0
/usr/local/lib/libjavatinyb.so
/home/pi/Gateway_2016/Intel/tinyb-master/build/java/jni/libjavatinyb.so.0.5.0
/home/pi/Gateway_2016/Intel/tinyb-master/build/java/jni/libjavatinyb.so.0
/home/pi/Gateway_2016/Intel/tinyb-master/build/java/jni/libjavatinyb.so
/home/pi/Gateway_2016/Intel/tinyb-master/build/src/libtinyb.so
/home/pi/Gateway_2016/Intel/tinyb-master/build/src/libtinyb.so.0
/home/pi/Gateway_2016/Intel/tinyb-master/build/src/libtinyb.so.0.5.0

The commands used to build tinyb are:
  mkdir build
  cd build
  cmake -DCMAKE_CXX_FLAGS=-std=c++11 -DBUILDJAVA=ON -DCMAKE_BUILD_TYPE=DEBUG ..
  make
  sudo make install 

Am I doing something wrong, or incomplete?  There were no errors reported by the cmake or make commands.

youcangetme, 2016-08-09:
Is your JAVA_HOME set to the JDK?

dfclark, 2016-08-09:
The app is being run on the RPi by Netbeans running on the Mac - the app is pushed down to the RPi and run there.  Netbeans has a Platform defined for the Remote Java SE, and in the platform settings there is a property known as java.home that is set to the JDK path (/usr/lib/jvm/jdk-8-oracle-arm32-vfp-hflt/jre).  The error message appears to be telling me that the libjavatinyb.so.0.5.0 library was found but that the embedded reference to libtinyb.so.0 is not resolved, even though libtinyb.so.0 is a peer of libjavatinyb.so.0.5.0 in /usr/local/lib.  Here's what /usr/local/lib looks like:

drwxrwsr-x  9 root staff    4096 Aug  8 16:50 .
drwxrwsr-x 11 root staff    4096 Aug  2 09:04 ..
drwxr-sr-x  2 root staff    4096 Aug  8 16:50 java
lrwxrwxrwx  1 root staff      17 Aug  8 11:19 libjavatinyb.so -> libjavatinyb.so.0
lrwxrwxrwx  1 root staff      21 Aug  8 11:19 libjavatinyb.so.0 -> libjavatinyb.so.0.5.0
-rw-r--r--  1 root staff 1291184 Aug  8 16:49 libjavatinyb.so.0.5.0
lrwxrwxrwx  1 root staff      13 Aug  8 11:19 libtinyb.so -> libtinyb.so.0
lrwxrwxrwx  1 root staff      17 Aug  8 11:19 libtinyb.so.0 -> libtinyb.so.0.5.0
-rw-r--r--  1 root staff 2363444 Aug  8 16:48 libtinyb.so.0.5.0

youcangetme, 2016-08-09:
I don't build off machine so I can't help you. I know on the Edison you had to do some extra tweaking to make it work with the Oracle JVM anyways.

petreeftime, 2016-08-09:
Hi,

I think that you need to add /usr/local/lib to PATH before running the Java program:
```export PATH=/usr/local/lib:$PATH``` 

Alternatively, install the library in /usr/lib instead of /usr/local/lib by adding:
```-DCMAKE_INSTALL_PREFIX=/usr```

dfclark, 2016-08-09:
Hi, and thanks for the response.  With a slight tweak, this worked!

The change for the install prefix resulted in the installation of the tinyb library files in 
/usr/lib/arm-linux-gnueabihf.

-- Installing: /usr/lib/arm-linux-gnueabihf/../lib/java/tinyb.jar
-- Installing: /usr/lib/arm-linux-gnueabihf/libjavatinyb.so.0.5.0
-- Installing: /usr/lib/arm-linux-gnueabihf/libjavatinyb.so.0
-- Installing: /usr/lib/arm-linux-gnueabihf/libjavatinyb.so
-- Removed runtime path from "/usr/lib/arm-linux-gnueabihf/libjavatinyb.so.0.5.0"
-- Installing: /usr/lib/arm-linux-gnueabihf/libtinyb.so.0.5.0
-- Installing: /usr/lib/arm-linux-gnueabihf/libtinyb.so.0
-- Installing: /usr/lib/arm-linux-gnueabihf/libtinyb.so

After the build finished I manually moved the libjavatinyb* and libtinyb* files up one level to /usr/lib, and then the app worked!  So there is a tweak needed still to fix the paths to get everything into /usr/lib and /usr/lib/java.

petreeftime, 2016-08-09:
It might work with the libraries in that folder as well. I'll close the bug now. :)

closed 2016-08-09 by petreeftime

dfclark, 2016-08-09:
I tried running the app on the RPi before moving the files, and the error was the same as before.  But after moving the libjavatinyb* and libtinyb* files the app worked!  Thanks again, for the library and the help!

d-f-clark, 2017-04-07:
When I build the TinyB library on my RaspberryPi it adds a bunch of characters to the names of the shared objects.  Here's an example:
Installing: /usr/lib/arm-linux-gnueabihf/libjavatinyb.so.0.5.0-8-g6e580f4.0.5.0-8-g6e580f4.0.5.0-8-g6e580f4
Where is the extra garbage ("-8-g6e580f4.") coming from, appended to the version number?  How do I suppress that?

downshifter, 2017-04-19:
This worked for me when I the same sort of problem running the tinyb.HelloTinyB:
export LD_LIBRARY_PATH=/usr/local/lib

Then to run the example:
java -classpath .:/usr/local/lib/java/tinyb.jar HelloTinyB hci0

With LD_LIBRARY_PATH set, -Djava.library.path=/usr/local/lib becomes unnecessary in the java command line