intel-iot-devkit/tinyb #42: Runtime error when attempting to write bytes to a characteristic

dfclark, 2016-08-23:
Running on Rasbian Jessie on RPi 3 Model B, with Bluez 5.41, Java build 1.8.0_65-b17.  
Getting this message:
Exception in thread "main" 
tinyb.BluetoothException: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method "WriteValue" with signature "ay" on interface "org.bluez.GattCharacteristic1" doesn't exist
	at tinyb.BluetoothGattCharacteristic.writeValue(Native Method)
	at hellotinyb.HelloTinyB.main(HelloTinyB.java:205)
#
# A fatal error has been detected by the Java Runtime Environment:
#

Built TinyB using these commands:
cd tinyb-master
sudo rm -r build
mkdir build
cd build
cmake -DCMAKE_CXX_FLAGS=-std=c++11 -DBUILDJAVA=ON -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_INSTALL_PREFIX=/usr ..
make
sudo make install

petreeftime, 2016-08-24:
What version of TinyB are you using? Only the latest upstream commit should be able to run on BlueZ 5.40+.

dfclark, 2016-08-24:
The try reported yesterday was done using a clone from 8/8/2016.  Today I cloned the current repository from https://github.com/intel-iot-devkit/tinyb.git, built and used that.  It worked.  Thanks for the prod, hadn't noticed the note for Bluez 5.41

petreeftime, 2016-08-25:
Great!

closed 2016-08-25 by petreeftime