intel-iot-devkit/tinyb #43: GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

mraugit, 2016-08-30:
I try to run the HelloTinyB example with the following configuration:

Platform: RaspberryPi 3 
Sensor: Simulation with Galaxy S6 edge and App "BLE Peripheral Simulator" (Heartbeat sensor)

cmake (ver 3.5.2)
glib/gio (ver. 2.40)

After some seconds I got the following error:

The discovery started: true
Address = 7E:EC:09:D8:DB:B1
 Name = Galaxy S6 edge
 Connected = false
Found device: 
Address = 7E:EC:09:D8:DB:B1
 Name = Galaxy S6 edge
 Connected = false
Sensor with the provided address connected
Services exposed by device:
(process:3890): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed
(process:3890): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

With gatttool the connection with the sensor simulation works fine. However I have to set the address-type to random, maybe this is the cause of the problem. I started gatttool with the following parameters:

gatttool -b 7E:EC:09:D8:DB:B1 -t random --interactive
...

petreeftime, 2016-08-31:
Hi! What version of TinyB are you running? You need to have commit 5d9cd1f to run on BlueZ 5.41.

I think the version of glib/gio you are running is a bit old, something in the API changed in version 2.42 or 2.44 and it made it partially incompatible with older versions. My recommendation would be to switch to 2.48 or later.

Also, step 4 in [Troubleshooting](https://github.com/intel-iot-devkit/tinyb/blob/5d9cd1f52eab6420c293677362c0fc05a2831032/TROUBLESHOOTING.md) might be related.

If these don't fix the problem, please let me know, I'll look into it.

mraugit, 2016-09-02:
Now it works. I use commit 5d9cd1f. The problem was the wrong bluez version. I have two bluez versions on my pi3. Version 5.23 in /usr/lib/bluetooth/bluetoothd (standard from pi-bluetooth package) and version 5.41 in /usr/libexec/bluetooth/bluetoothd (own build).  

The change of the following line in /lib/systemd/system/bluetooth.service fixes the problem:

Old: ExecStart=/usr/lib/bluetooth/bluetoothd
New: ExecStart=/usr/libexec/bluetooth/bluetoothd

Thanks

closed 2016-09-02 by mraugit