intel-iot-devkit/tinyb #45: How to access advertisements, and manufacturer-specific data in advertisements

dfclark, 2016-09-02:
I have an app running on Raspbian that is interacting with my BLE devices, can connect and enumerate services and characteristics.  But before connecting I need to be able to access the mfg-specific data in advertisements, and can't see how to do that.  Is this available, and could you provide any pointers on how to do it?   Thanks

petreeftime, 2016-09-02:
Hi! My best bet is that you need to read the ManufacturerData property of the device. It's mentioned in the [BlueZ Device API](http://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/device-api.txt), but missing from TinyB right now, I think I might've missed the change that added it.

dfclark, 2016-09-02:
Is there any possibility that you will be adding this support soon?  If not, are there any sort of guidelines for how to update the code myself?  It's been over 10 years since I've done any JNI work, but it looks like I'd have to add a JNI method, plus some C code, to expose that BlueZ property using a HashMap in Java and a std::unordered_map in c++?

petreeftime, 2016-09-05:
The way to update the code is regenerate the gdbus code with gdbus_codegen after updating the org.bluez.xml file. But yes, I'll add support.

dfclark, 2016-09-07:
Not wanting to be pushy, but do you know when that capability might be added?  And there may actually be other behavior that is a critical need.  In the IoT's the apps will most likely want to react to devices that come into range, and without polling the only way to know this is to see the advertisements.  Many IoT devices put critical information in the advertisements - in fact, for some devices, there is no need to connect as their total value is the data that is advertised.  Take a temperature sensor for example - a basic sensor only needs to advertise its current reading, in most cases there should be no need to connect to read a characteristic.  So probably what's needed in TinyB is the ability for the client to register for callbacks with the BluetoothManager, probably in conjunction with the call to startDiscovery().  Thanks for listening.

petreeftime, 2016-09-07:
Hi! I was just working on this, I can't say for sure when it will be done, but I think Friday should be a good estimate.

You can already use manager->find with callbacks, and they will trigger when a new device is discovered, albeit from C++ only right now. Could you please add another issue to remind me to add Java support for this?

petreeftime, 2016-09-08:
I'm having some issues finding a device which actually has the Manufacturer data field set, so it might take a while longer until I find one or figure out how to set it on a device.

dfclark, 2016-09-08:
If push comes to shove I can send you one of mine.  Or perhaps I could somehow do some testing for you?

ccustine, 2016-09-09:
TI Sensortag stock firmware has a couple of items in Mfg data, if you happen to have one available.

petreeftime, 2016-09-09:
I have a TI Sensortag, but BlueZ does not seem to populate anything in ManufacturerData. I'll add a patch to a secondary branch, maybe you can test if it works.

petreeftime, 2016-09-09:
I commited my code to the https://github.com/intel-iot-devkit/tinyb/tree/mfg-data branch, if you compile and run the list_mfg executable, it should try listing the data in the ManufacturerData property. For me each device returns nothing.

I've also used ```gdbus introspection``` command to confirm that the property is indeed empty.

dfclark, 2016-09-09:
Thank you for the update, I'll check it out and let you know what happens with my devices.

ccustine, 2016-09-09:
I will be able to test this soon and will let you know as well.

dfclark, 2016-09-09:
I've been unable to get the tinyb library to load, after running the build instructions.  Here's the contents of the shell script I ran to build the API:
pi@raspberrypi:~/Gateway_2016/Intel/tinyb-mfg-data/tinyb $ cat build_tinyb.sh 
  sudo rm -r build
  mkdir build
  cd build
  cmake -DCMAKE_CXX_FLAGS=-std=c++11 -DBUILDJAVA=ON -DCMAKE_BUILD_TYPE=DEBUG ..
  make
  sudo make install
pi@raspberrypi:~/Gateway_2016/Intel/tinyb-mfg-data/tinyb $ 

Here's the last output from the build, followed by a listing showing the contents of usr/local/lib, then the output from attempting to run the app:
Install the project...
-- Install configuration: "DEBUG"
-- Installing: /usr/local/lib/../lib/java/tinyb.jar
-- Installing: /usr/local/lib/libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
-- Installing: /usr/local/lib/libjavatinyb.so.0.5.0-4-g23b0754
-- Installing: /usr/local/lib/libjavatinyb.so
-- Removed runtime path from "/usr/local/lib/libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754"
-- Up-to-date: /usr/local/include/tinyb.hpp
-- Installing: /usr/local/include/tinyb
-- Up-to-date: /usr/local/include/tinyb/BluetoothManager.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothObject.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothAdapter.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothGattCharacteristic.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothEvent.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothDevice.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothException.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothGattService.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothUUID.hpp
-- Up-to-date: /usr/local/include/tinyb/BluetoothGattDescriptor.hpp
-- Installing: /usr/local/lib/pkgconfig/tinyb.pc
-- Installing: /usr/local/lib/libtinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
-- Installing: /usr/local/lib/libtinyb.so.0.5.0-4-g23b0754
-- Installing: /usr/local/lib/libtinyb.so

pi@raspberrypi:/usr/local/lib $ pwd
/usr/local/lib
pi@raspberrypi:/usr/local/lib $ ls -al *tinyb*
lrwxrwxrwx 1 root staff      32 Sep  9 15:19 libjavatinyb.so -> libjavatinyb.so.0.5.0-4-g23b0754
lrwxrwxrwx 1 root staff      66 Sep  9 15:19 libjavatinyb.so.0.5.0-4-g23b0754 -> libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
-rw-r--r-- 1 root staff 1291416 Sep  9 15:18 libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
lrwxrwxrwx 1 root staff      28 Sep  9 15:19 libtinyb.so -> libtinyb.so.0.5.0-4-g23b0754
lrwxrwxrwx 1 root staff      62 Sep  9 15:19 libtinyb.so.0.5.0-4-g23b0754 -> libtinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
-rw-r--r-- 1 root staff 2462532 Sep  9 15:18 libtinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754

pi@raspberrypi:~/NetBeansProjects/HelloTinyB $ java -Djava.library.path=/usr/local/lib/:/usr/local/lib/java/  -jar /home/pi/NetBeansProjects/HelloTinyB/dist/HelloTinyB.jar E3:F0:66:B6:02:94
15:27:43.355 [main] TRACE hellotinyb.HelloTinyB - Device address is E3:F0:66:B6:02:94
Native code library failed to load.
java.lang.UnsatisfiedLinkError: /usr/local/lib/libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754: libtinyb.so.0.5.0-4-g23b0754: cannot open shared object file: No such file or directory
pi@raspberrypi:~/NetBeansProjects/HelloTinyB $

petreeftime, 2016-09-12:
Add ```-DCMAKE_INSTALL_PREFIX=/usr``` to the cmake command, as it installs the files in /usr/local/lib instead of /usr/lib

petreeftime, 2016-09-12:
I managed to find a device which has ManufacturerData set and I fixed the function. I'll add Java binding and the rest of the missing functions and push to the master branch.

dfclark, 2016-09-13:
Regarding the suggestion to use -DCMAKE_INSTALL_PREFIX:  I did that originally and the process creates another folder in the hierarchy called "arm-linux-gnueabihf", and the end result was unresolved references when TinyB is accessed in the application.  So I removed that -DCMAKE... option, let the components install into /usr/local/lib, and that worked.  With the latest clone that same script produces the issues reported above.  Adding the -DCMAKE... option back in produces the original issue.  Here's the latest output, using the -DCMAKE... argument:
Install the project...
-- Install configuration: "DEBUG"
-- Installing: /usr/lib/arm-linux-gnueabihf/../lib/java/tinyb.jar
-- Installing: /usr/lib/arm-linux-gnueabihf/libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
-- Installing: /usr/lib/arm-linux-gnueabihf/libjavatinyb.so.0.5.0-4-g23b0754
-- Installing: /usr/lib/arm-linux-gnueabihf/libjavatinyb.so
-- Removed runtime path from "/usr/lib/arm-linux-gnueabihf/libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754"
-- Installing: /usr/include/tinyb.hpp
-- Installing: /usr/include/tinyb
-- Installing: /usr/include/tinyb/BluetoothManager.hpp
-- Installing: /usr/include/tinyb/BluetoothObject.hpp
-- Installing: /usr/include/tinyb/BluetoothAdapter.hpp
-- Installing: /usr/include/tinyb/BluetoothGattCharacteristic.hpp
-- Installing: /usr/include/tinyb/BluetoothEvent.hpp
-- Installing: /usr/include/tinyb/BluetoothDevice.hpp
-- Installing: /usr/include/tinyb/BluetoothException.hpp
-- Installing: /usr/include/tinyb/BluetoothGattService.hpp
-- Installing: /usr/include/tinyb/BluetoothUUID.hpp
-- Installing: /usr/include/tinyb/BluetoothGattDescriptor.hpp
-- Installing: /usr/lib/pkgconfig/tinyb.pc
-- Installing: /usr/lib/arm-linux-gnueabihf/libtinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
-- Installing: /usr/lib/arm-linux-gnueabihf/libtinyb.so.0.5.0-4-g23b0754
-- Installing: /usr/lib/arm-linux-gnueabihf/libtinyb.so

If the goal is to place the runtime files into the /usr/lib folder then there will be another step necessary to move the files back to that folder from the /usr/lib/arm-linux-gnueabihf folder.  Originally I did that as well, and originally that worked.  But now it doesn't.

I have noticed that the output from the new build produces a set of files different from the set formerly produced from the master branch.  The file names are of course different, but the actual set of files is different as well.  Here's the list from the former master build:

pi@raspberrypi:/usr/local/lib $ ls -al *tinyb*
lrwxrwxrwx 1 root staff      32 Aug 24 11:06 libjavatinyb.so -> libjavatinyb.so.0.5.0-2-g5d9cd1f
lrwxrwxrwx 1 root staff      21 Aug  8 11:19 libjavatinyb.so.0 -> libjavatinyb.so.0.5.0
-rw-r--r-- 1 root staff 1291184 Aug  8 16:49 libjavatinyb.so.0.5.0
lrwxrwxrwx 1 root staff      66 Aug 24 11:06 libjavatinyb.so.0.5.0-2-g5d9cd1f -> libjavatinyb.so.0.5.0-2-g5d9cd1f.0.5.0-2-g5d9cd1f.0.5.0-2-g5d9cd1f
-rw-r--r-- 1 root staff 1291184 Aug 24 11:05 libjavatinyb.so.0.5.0-2-g5d9cd1f.0.5.0-2-g5d9cd1f.0.5.0-2-g5d9cd1f
lrwxrwxrwx 1 root staff      28 Aug 24 11:06 libtinyb.so -> libtinyb.so.0.5.0-2-g5d9cd1f
lrwxrwxrwx 1 root staff      17 Aug  8 11:19 libtinyb.so.0 -> libtinyb.so.0.5.0
-rw-r--r-- 1 root staff 2363444 Aug  8 16:48 libtinyb.so.0.5.0
lrwxrwxrwx 1 root staff      62 Aug 24 11:06 libtinyb.so.0.5.0-2-g5d9cd1f -> libtinyb.so.0.5.0-2-g5d9cd1f.0.5.0-2-g5d9cd1f.0.5.0-2-g5d9cd1f
-rw-r--r-- 1 root staff 2368624 Aug 24 11:04 libtinyb.so.0.5.0-2-g5d9cd1f.0.5.0-2-g5d9cd1f.0.5.0-2-g5d9cd1f

Here's the set produced from the latest branch:
pi@raspberrypi:/usr/local/lib $ ls -al *tinyb*
lrwxrwxrwx 1 root staff      32 Sep  9 15:19 libjavatinyb.so -> libjavatinyb.so.0.5.0-4-g23b0754
lrwxrwxrwx 1 root staff      66 Sep  9 15:19 libjavatinyb.so.0.5.0-4-g23b0754 -> libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
-rw-r--r-- 1 root staff 1291416 Sep  9 15:18 libjavatinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
lrwxrwxrwx 1 root staff      28 Sep  9 15:19 libtinyb.so -> libtinyb.so.0.5.0-4-g23b0754
lrwxrwxrwx 1 root staff      62 Sep  9 15:19 libtinyb.so.0.5.0-4-g23b0754 -> libtinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754
-rw-r--r-- 1 root staff 2462532 Sep  9 15:18 libtinyb.so.0.5.0-4-g23b0754.0.5.0-4-g23b0754.0.5.0-4-g23b0754

petreeftime, 2016-09-13:
It does seem to have this issue for me too, I ran with ```export LD_LIBRARY_PATH=/usr/local/lib64:$LD_LIBRARY_PATH``` and it worked.

petreeftime, 2016-09-16:
This feature is not up on the master branch, including Java bindings.

closed 2016-09-16 by petreeftime

dfclark, 2016-09-21:
Hi Petre.  I've updated to the latest from master, and now accessing the manufacturer's data is simple - I call getManufacturerData() and get a Map of advertisements.  Thank you!

But of course I have a question or two.
1) Since I'm dealing with a given device, and I'm calling getManufacturerData(), why is the returned value a Map with the entries keyed by the mfg ID?  There should be one Manufacturer for a given device, but of course there could/should be multiple advertisements - is the Map a list of recent advertisements, and if so, what is the ordering?
2) There is still a need to get callbacks for advertisements - what would be very handy would be the ability to register for callbacks for advertisements for a given MAC address.  Ideally this would be exposed to Java, and would provide a byte array that is the advertisement data.

Thanks.

petreeftime, 2016-09-22:
Hi,

1) That's how BlueZ returns the data, and since I only have one device that has MFG data set, I expected that some device might have more than one MFG ID.
2) I'll look into adding callbacks or searching through MFGs. Would being able to have a callback for ManufacturerData updates per BluetoothDevice be helpful?

dfclark, 2016-09-22:
Hi Petre.  Ideally the app would be able to register (and deregister) for callbacks when advertisements are received, which would allow the app to react to these temporally.  For example, I have a sensor that transmits its data via the advertisement - because it posts advertisements much more frequently than it takes a sensor reading there is a toggle bit in the advertisement that is flipped when the sensor transmits a new value.  The app should be able see every advertisement, take note of this bit, and only process the sensor data when the bit is flipped.  Without the callback mechanism the app must poll the list of devices maintained by the BluetoothManager, and then examine the advertisement data attached to each BluetoothDevice object.  This is almost workable except that the polling frequency would have to be some set value, rather than being able to react to the asynchronous arrival of the advertisements from multiple sensor devices.