intel-iot-devkit/tinyb #47: TinyB can't find services

AxelUlmestig, 2016-09-27:
I'm running the HelloTinyB example to connect to Intel Edisons using BLE. I get to the "Services exposed by device:" part and then it doesn't find anything and loops forever.

```
java -cp tinyb/build/examples/java/HelloTinyB.jar:/usr/lib/java/tinyb.jar HelloTinyB "XX:XX:XX:XX:XX:XX"
The discovery started: true
Address = YY:YY:YY:YY:YY:YY Name = W Connected = false
Error: GDBus.Error:org.bluez.Error.InProgress: Operation already in progress
Found device: Address = XX:XX:XX:XX:XX:XX Name = EdisonDevice2 Connected = false
Sensor with the provided address connected
Services exposed by device:
```

I'm able to connect and see the services and characteristics of the other Edison device using the BLE Scanner android app.

I've diverted slightly from the example as I'm using build/examples/java/HelloTinyB.jar instead of examples/java/HelloTinyB.jar because that one wasn't available after I built it like this:

```
mkdir build
cd build
cmake .. -DBUILDJAVA=ON -DCMAKE_INSTALLPREFIX=/usr
make
make install
```

Otherwise I've just replaced the mac address and the service uuid with my own.

The "Error: GDBus.Error:org.bluez.Error.InProgress: Operation already in progress" output appears sometimes but not always. But the result is always the same.

**Other info**

```
root@edison:~# ls -la /usr/lib/*tinyb.so
lrwxrwxrwx 1 root root 17 Jun  7 00:50 /usr/lib/libjavatinyb.so -> libjavatinyb.so.0
lrwxrwxrwx 1 root root 13 Jun  7 00:50 /usr/lib/libtinyb.so -> libtinyb.so.0
```

```
root@edison:~# cat /etc/dbus-1/system.d/bluetooth.conf
<!-- This configuration file specifies the required security policies
     for Bluetooth core daemon to work. -->

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- ../system.conf have denied everything, so we just punch some holes -->

  <policy user="root">
    <allow own="org.bluez"/>
    <allow send_destination="org.bluez"/>
    <allow send_interface="org.bluez.Agent"/>
    <allow send_interface="org.bluez.HandsfreeAgent"/>
    <allow send_interface="org.bluez.MediaEndpoint"/>
    <allow send_interface="org.bluez.MediaPlayer"/>
  </policy>

  <policy context="default">
    <allow own="org.bluez"/>
    <allow own="org.bluez.obex"/>
    <allow send_destination="org.bluez"/>
    <allow send_destination="org.bluez.obex"/>
    <allow send_interface="org.bluez.Agent1"/>
    <allow send_interface="org.bluez.MediaEndpoint1"/>
    <allow send_interface="org.bluez.Profile1"/>
    <allow send_interface="org.bluez.ThermometerWatcher1"/>
    <allow send_interface="org.bluez.AlertAgent1"/>
    <allow send_interface="org.bluez.GattCharacteristic1"/>
    <allow send_interface="org.bluez.GattDescriptor1"/>
    <allow send_interface="org.bluez.GattService1"/>
    <allow send_interface="org.bluez.HeartRateWatcher1"/>
    <allow send_interface="org.bluez.CyclingSpeedWatcher1"/>
  </policy>

</busconfig>
```

petreeftime, 2016-09-27:
Hi, can you run ```ps aux | grep bluetoothd``` and confirm that it's running with the -E flag?

AxelUlmestig, 2016-09-27:
```
root@edison:~# ps | grep bluetoothd
 1217 root      4928 S    /usr/lib/bluez5/bluetooth/bluetoothd -E
 1780 root      2656 S    grep bluetoothd
```

petreeftime, 2016-09-27:
Did you edit the bluetooth.conf file by hand? Did you try restarting the board to reload the dbus rules?

AxelUlmestig, 2016-09-27:
The -E tag was added automatically to the bluetooth.conf. I tried replacing it with --experimental as someone mentioned in another thread and I've since returned it to -E.

I've restarted both the boards a many times. It seems that I can't connect with my phone after I've tried connecting with the Edison. I also got a new error when I tried to run it just now:

```
java -cp tinyb/build/examples/java/HelloTinyB.jar:/usr/lib/java/tinyb.jar HelloTinyB "98:4F:EE:03:5C:FC"
The discovery started: true
Address = XX:XX:XX:XX:XX:XX Name = EdisonDevice2 Connected = false
Error: GDBus.Error:org.bluez.Error.InProgress: Operation already in progress
Found device: Address = XX:XX:XX:XX:XX:XX Name = EdisonDevice2 Connected = false
Sensor with the provided address connected
Services exposed by device:

(process:1837): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed
Exception in thread "main" java.lang.NoSuchMethodError: tinyb.BluetoothGattService.getUUID()Ljava/lang/String;
        at HelloTinyB.getService(HelloTinyB.java:67)
        at HelloTinyB.main(HelloTinyB.java:159)
```

petreeftime, 2016-09-27:
I noticed a typo in your cmake command, it should be ```-DCMAKE_INSTALL_PREFIX=/usr``` instead of ```-DCMAKE_INSTALLPREFIX=/usr```

AxelUlmestig, 2016-09-27:
I rebuilt it with the correct flag now. It's giving me some new error messages now which might be useful:

```
root@edison:~/tinyb# java -cp tinyb/build/examples/java/HelloTinyB.jar:/usr/lib/java/tinyb.jar HelloTinyB "98:4F:EE:03:5C:FC"

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed

(process:407): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed
The discovery started: true
Address = 98:4F:EE:03:5C:FC Name = edison10 Connected = false
Error: GDBus.Error:org.bluez.Error.InProgress: Operation already in progress
Found device: Address = 98:4F:EE:03:5C:FC Name = edison10 Connected = false
Sensor with the provided address connected
Services exposed by device:
Exception in thread "main" java.lang.NoSuchMethodError: tinyb.BluetoothGattService.getUUID()Ljava/lang/String;
        at HelloTinyB.getService(HelloTinyB.java:67)
        at HelloTinyB.main(HelloTinyB.java:159)
```

AxelUlmestig, 2016-09-27:
On boot I run the following commands:
```
rfkill unblock bluetooth
killall bluetoothd
hciconfig hci0 up
```
Is there perhaps something that needs to be enabled?

petreeftime, 2016-09-27:
Can you please tell what version of BlueZ, GLib and TinyB you are trying to run?

AxelUlmestig, 2016-09-27:
I reflashed the edison today from the latest available image.
In the configuration tool it says this:
"Firmware version installed: 201606061707."

I have not made any updated to any updated to Bluez or Glib.
I cloned the TinyB repo early last week. 

How can I check the versions of glib and bluez?

petreeftime, 2016-09-27:
```bluetoothctl --version``` should give you the BlueZ version and ``` ls /usr/lib | grep glib ``` should show what version of GLib is installed.

It seems to be loading an older version of the library, because ```Exception in thread "main" java.lang.NoSuchMethodError: tinyb.BluetoothGattService.getUUID()Ljava/lang/String;``` was changed a few months ago.

I think you might need to run ```cmake -DBUILDJAVA=ON -DCMAKE_INSTALLPREFIX=/usr ..```, with the folder at the end instead of at the beginning. What was the output of the make install command?

AxelUlmestig, 2016-09-27:
```
root@edison:~# bluetoothctl --version
5.37
```
```
root@edison:~# ls /usr/lib/ | grep glib
glib-2.0
libdbus-glib-1.la
libdbus-glib-1.so
libdbus-glib-1.so.2
libdbus-glib-1.so.2.2.2
libglib-2.0.la
libglib-2.0.so
libglib-2.0.so.0
libglib-2.0.so.0.4000.0
libpyglib-2.0-python.la
libpyglib-2.0-python.so
libpyglib-2.0-python.so.0
libpyglib-2.0-python.so.0.0.0
```
**And here's the complete log from the build:**
```
root@edison:~/tinyb/build# cmake -DBUILDJAVA=ON -DCMAKE_INSTALL_PREFIX=/usr ..
INFO - libtinyb Version 0.5.0-4-g22c1af1
-- JNI_INCLUDE_DIRS=/usr/lib/jvm/java-8-openjdk/include;/usr/lib/jvm/java-8-openjdk/include/linux;/usr/lib/jvm/java-8-openjdk/include
-- JNI_LIBRARIES=/usr/lib/jvm/java-8-openjdk/jre/lib/i386/libjawt.so;/usr/lib/jvm/java-8-openjdk/jre/lib/i386/libjvm.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/root/tinyb
root@edison:~/tinyb/build# make
Generating JNI headers..
[  6%] Built target tinybjar
[ 32%] Built target tinyb
[ 54%] Built target javatinyb
[ 60%] Built target AsyncTinyB
[ 66%] Built target HelloTinyB
[ 72%] Built target Notification
[ 76%] Built target asynctinyb
[ 80%] Built target checkinit
[ 84%] Built target esstinyb
[ 88%] Built target hellotinyb
[ 92%] Built target list_mfg
[ 96%] Built target notifications
[100%] Built target uuid
root@edison:~/tinyb/build# make install
Generating JNI headers..
[  6%] Built target tinybjar
[ 32%] Built target tinyb
[ 54%] Built target javatinyb
[ 60%] Built target AsyncTinyB
[ 66%] Built target HelloTinyB
[ 72%] Built target Notification
[ 76%] Built target asynctinyb
[ 80%] Built target checkinit
[ 84%] Built target esstinyb
[ 88%] Built target hellotinyb
[ 92%] Built target list_mfg
[ 96%] Built target notifications
[100%] Built target uuid
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/../lib/java/tinyb.jar
-- Installing: /usr/local/lib/libjavatinyb.so.0.5.0-4-g22c1af1.0.5.0-4-g22c1af1.0.5.0-4-g22c1af1
-- Installing: /usr/local/lib/libjavatinyb.so.0.5.0-4-g22c1af1
-- Installing: /usr/local/lib/libjavatinyb.so
-- Removed runtime path from "/usr/local/lib/libjavatinyb.so.0.5.0-4-g22c1af1.0.5.0-4-g22c1af1.0.5.0-4-g22c1af1"
-- Installing: /usr/local/include/tinyb.hpp
-- Installing: /usr/local/include/tinyb
-- Installing: /usr/local/include/tinyb/BluetoothGattService.hpp
-- Installing: /usr/local/include/tinyb/BluetoothUUID.hpp
-- Installing: /usr/local/include/tinyb/BluetoothGattCharacteristic.hpp
-- Installing: /usr/local/include/tinyb/BluetoothGattDescriptor.hpp
-- Installing: /usr/local/include/tinyb/BluetoothEvent.hpp
-- Installing: /usr/local/include/tinyb/BluetoothObject.hpp
-- Installing: /usr/local/include/tinyb/BluetoothException.hpp
-- Installing: /usr/local/include/tinyb/BluetoothManager.hpp
-- Installing: /usr/local/include/tinyb/BluetoothAdapter.hpp
-- Installing: /usr/local/include/tinyb/BluetoothDevice.hpp
-- Installing: /usr/local/lib/pkgconfig/tinyb.pc
-- Installing: /usr/local/lib/libtinyb.so.0.5.0-4-g22c1af1.0.5.0-4-g22c1af1.0.5.0-4-g22c1af1
-- Installing: /usr/local/lib/libtinyb.so.0.5.0-4-g22c1af1
-- Installing: /usr/local/lib/libtinyb.so

```

petreeftime, 2016-09-27:
The issue seems to be that it's installing to /usr/local/lib instead of /usr/lib. Can you please move the files from /usr/local/lib to /usr/lib? It should most likely solve the problems.

petreeftime, 2016-09-27:
Ok, I've noticed that both your BlueZ and Glib is out of date, I'll see if I can do something to fix that or if I can find someone that can provide you with updated packages.

You'll have to checkout tag 0.5.0 to use TinyB with BlueZ 5.37 until this is done. I'm not sure if the Glib errors will still be there. If you need the latest features I recommend updating BlueZ to 5.42 and glib to 2.48 until official packages are available (it might take a while).

I'll push a patch that fixes the GDBus error at the end.

AxelUlmestig, 2016-09-28:
Thank you very much.
I will try updating BlueZ and GLib

AxelUlmestig, 2016-09-28:
Hi, I tried checking out release 0.5 of TinyB. Now it found the service and the characteristic. But I ran an into another problem.

```
root@edison:~# java -cp tinyb/build/examples/java/HelloTinyB.jar:/usr/lib/java/tinyb.jar HelloTinyB "98:4F:EE:03:5C:FC"
The discovery started: true
Address = 98:4F:EE:03:5C:FC Name = edison10 Connected = false
Found device: Address = 98:4F:EE:03:5C:FC Name = edison10 Connected = false
Sensor with the provided address connected
Services exposed by device:
UUID: 0000180a-0000-1000-8000-00805f9b34fb
UUID: 0000181a-0000-1000-8000-00805f9b34fb
Found service 0000181a-0000-1000-8000-00805f9b34fb
Found the temperature characteristics
Exception in thread "main" tinyb.BluetoothException: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method "ReadValue" with signature "a{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist

        at tinyb.BluetoothGattCharacteristic.readValue(Native Method)
        at HelloTinyB.main(HelloTinyB.java:197)
```
The error occurs on this line:
```
        while (running) {
            byte[] tempRaw = tempValue.readValue(); //here is where it crashes
            System.out.print("Temp raw = {");
            for (byte b : tempRaw) {
                System.out.print(String.format("%02x,", b));
            }
            System.out.print("}");
```
I'm gonna try to update BlueZ and see if gets resolved

petreeftime, 2016-09-28:
Hi, that error is caused by having BlueZ 5.40+ with TinyB 0.5 or older. The patch that fixes the issue is https://github.com/intel-iot-devkit/tinyb/commit/5d9cd1f52eab6420c293677362c0fc05a2831032.

AxelUlmestig, 2016-09-28:
Hi, I upgraded to BlueZ 5.42 and ran with the latest version of TinyB. It works now (!), but there seems to be some trouble with the disconnecting. 

I added a loop counter in the while loop at the end of the example to make sure it reaches the sensor.disconnect() line. The loop exits after three iterations.

```
root@edison0:~# java -cp tinyb/build/examples/java/HelloTinyB.jar:/usr/lib/java/tinyb.jar HelloTinyB "XX:XX:XX:XX:XX:XX"
The discovery started: true
Address = XX:XX:XX:XX:XX:XX Name = edison10 Connected = false
Found device: Address = XX:XX:XX:XX:XX:XX Name = edison10 Connected = false
Sensor with the provided address connected
Services exposed by device:
UUID: 00001801-0000-1000-8000-00805f9b34fb
UUID: 0000180a-0000-1000-8000-00805f9b34fb
UUID: 0000181a-0000-1000-8000-00805f9b34fb
Found service 0000181a-0000-1000-8000-00805f9b34fb
Found the temperature characteristics
Temp raw = {32,35,2e,32,30,} Temp: Object = 106.390625C, Ambient = 100.359375C
Temp raw = {32,35,2e,32,30,} Temp: Object = 106.390625C, Ambient = 100.359375C
Temp raw = {32,35,2e,32,30,} Temp: Object = 106.390625C, Ambient = 100.359375C
```
Then I tried to run the exact same command again afterwards:
```
root@edison0:~# java -cp tinyb/build/examples/java/HelloTinyB.jar:/usr/lib/java/tinyb.jar HelloTinyB "XX:XX:XX:XX:XX:XX"
The discovery started: true
Address = XX:XX:XX:XX:XX:XX Name = edison10 Connected = false
Address =YY:YY:YY:YY:YY:YY Name = W Connected = false
Exception in thread "main" tinyb.BluetoothException: GDBus.Error:org.bluez.Error.InProgress: Operation already in progress
        at tinyb.BluetoothManager.stopDiscovery(Native Method)
        at HelloTinyB.main(HelloTinyB.java:123)
```

petreeftime, 2016-09-28:
I pushed a fix for that: https://github.com/intel-iot-devkit/tinyb/commit/d9c777bc576b5365c2064d55fe872f272faa92e8.

AxelUlmestig, 2016-09-28:
That solved it! Thank you so much for taking the time to help me.

**To anyone else coming to this thread.** 
I think the original problems were that build had to be run with these flags:
```
mkdir build
cd build
cmake .. -DBUILDJAVA=ON -DCMAKE_INSTALLPREFIX=/usr
make
make install
```
And BlueZ had to be updated to a newer version (5.42 in my case).
BlueZ 5.42 can be found here:
http://www.bluez.org/release-of-bluez-5-42/

closed 2016-09-28 by AxelUlmestig