intel-iot-devkit/tinyb #48: How to build this code base in windows.. looking for a java based project..

krrishnayak, 2016-10-06:


petreeftime, 2016-10-06:
Hi! Do you want to build it on Windows or run it on Windows? Unfortunately running it on Windows is not really an option at this time and it's very unlikely that it will be possible to do this soon.

krrishnayak, 2016-10-06:
hello.. i want to build it in windows and run it as java application. I am not clear with the CMake steps to build this project in windows.

petreeftime, 2016-10-06:
I will look into building it on Windows, although it might be fairly complicated, and it's not going to be a priority. But if you want to run in under Windows, it will not work. It should be much easier to just build in on the Linux machine where you will run it. It relies on BlueZ which only works on Linux.

krrishnayak, 2016-10-07:
can u list down the commands to build it in linux using cmake

petreeftime, 2016-10-07:
You can find information about building in the README.

gustavo84, 2016-11-22:
I've tried it, tinyb in ubuntu16 and run correctly. But would be wonderfull that tinyb run on windows.
¿Its more dificult? You think that our library its the first on this universe that run on java for bluetooth le with gatt services. And in few months you will have all the market

gustavo84, 2017-01-04:
I would like to contribute this functionality it would be most helpful. 
There is a driver on C, compatible to three platform (windows,linux,solaris).
I think that is posible develop jni connectors and i would like to contribute. https://github.com/NordicSemiconductor/pc-ble-driver