intel-iot-devkit/tinyb #51: tinyb does not show all available characteristics

mraugit, 2016-10-12:
I try to enable the notification of the movement sensor of a TI Simplelink Sensor TAG ([http://www.ti.com/ww/en/wireless_connectivity/sensortag2015/])

The full GATT-Table could be found at the end of this website:
[http://www.ti.com/ww/en/wireless_connectivity/sensortag2015/tearDown.html#main]

With the gatttool command line I can enable the notification by writing a 01:00 to the characteristic with the handle 0x3A (characteristic type is 0x2902, Client Characteristic Configuration). 

When I try to find the characteristic with tinyb the find function doesn't came back. When I try to write to the corresponding gatt descriptor I got a "permission denied" error.

The command line user and the user of the java programe is the same.

Why tinyb does not find all available characteristics?

petreeftime, 2016-10-12:
Hi!

What version of TinyB and BlueZ are you running? Did you follow the steps in [TROUBLESHOOTING.md](https://github.com/intel-iot-devkit/tinyb/blob/master/TROUBLESHOOTING.md)?
Can you run the hellotinyb  example (the native one, not the java one) and see if any Gatt Service or Characteristic is listed?

mraugit, 2016-10-12:
I use TinyB commit 5d9cd1f and BlueZ 5.41.
I've checked all possible issues described in TROUBLESHOOTING.md, but all these things are ok.

I changed the hellotinyb.cpp file as follows:
[hellotinyb.cpp.txt](https://github.com/intel-iot-devkit/tinyb/files/525385/hellotinyb.cpp.txt)

After running the changed example I got the following output:
[log.txt](https://github.com/intel-iot-devkit/tinyb/files/525399/log.txt)

With gatttool I can find and write to the corresponding characteristic:

[A0:E6:F8:B7:16:87][LE]> char-desc
...
handle: 0x0037, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x0038, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x0039, uuid: f000aa81-0451-4000-b000-000000000000
**handle: 0x003a, uuid: 00002902-0000-1000-8000-00805f9b34fb**
handle: 0x003b, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x003c, uuid: f000aa82-0451-4000-b000-000000000000
handle: 0x003d, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x003e, uuid: f000aa83-0451-4000-b000-000000000000
...

petreeftime, 2016-10-12:
Hi!

I don't think you need to explicitly write 1 to that characteristic. Just use ```enable_value_notifications``` like in the [notifications example](https://github.com/intel-iot-devkit/tinyb/blob/master/examples/notifications.cpp#L140) (or [for Java](https://github.com/intel-iot-devkit/tinyb/blob/master/examples/java/Notification.java#L193)).

Tell me if this helps.

mraugit, 2016-10-13:
This was also my expectation. The notification works for temperature, humidity, barometric pressure but not for the movement sensor. When I send 01:00 to the characteristic 00002902-0000-1000-8000-00805f9b34fb by command line and start afterwards my java program, the notification for movement works fine. But after a restart of the system the notification of the movement sensor doesn't work again.

dbonino, 2016-10-13:
I actually have a similar issue. characteristic with uuid: 00002902-0000-1000-8000-00805f9b34fb cannot be written, and the same happens for the I/O configuration characteristic of the sensor tag: f000aa66-0451-4000-b000-000000000000.
However notification works for every characteristic, including movement.

mraugit, 2016-10-13:
Hi dbonino,
do you use the same sensor tag like me?
TI Simplelink Sensor TAG (http://www.ti.com/ww/en/wireless_connectivity/sensortag2015/)

dbonino, 2016-10-14:
> sensortag2015/)

Yes, indeed

juiice, 2016-10-17:
Hi,
I had the same problem with a Polar H7 heart rate sensor.
But looking more carefully the "org.bluetooth.service.heart_rate" (uuid : 0x180D) -> https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.heart_rate.xml&u=org.bluetooth.service.heart_rate.xml
we can see "org.bluetooth.descriptor.gatt.client_characteristic_configuration" (uuid 0x2902) is a "descriptor" of the characteristc "org.bluetooth.characteristic.heart_rate_measurement" (uuid 0x2a37)

Now I am able to find the configuration "descriptor" from the "characteristic" (and not from the "service" as I tried first). I still have a problem of write permission in this descriptor, but I will open a new issue if I do not find a solution.

mraugit, 2016-10-17:
Hi dbonino,
I think I've found the cause of the problem with the TI Simplelink sensor. If I configure the interval period of the movement sensor larger than, 0x64 the notification does not work. If I choose a period less than or equal 0x64 the notification works fine. Can you please check this with your sensor.

Regardless of this, it should be possible to access characteristics of type 0x2902 (Client Characteristic Configuration). With gatttool it is possible to read-from or write-to this kind of characteristics. Tinyb only finds characteristics like the gatttool command "characteristics", but the gatttool command "char-desc" finds a lot more characteristics. I suggest to change this issue from bug to enhancement, what do you think?

mraugit, 2016-10-17:
Hi juiice,

I also have a problem with the write permission if I try to access the 0x2902 characteristic with the descriptor object.

dbonino, 2016-10-18:
Hi mraugit,
I'll try your settings for the notification period and see if going over 0x64 stops the tag from sending notifications. By the way, have you checked the maximum allowed period on the Sensor Tag characteristic documentation?

For what concerns the ability to write the 0x2902 characteristic, is fine for me to move it from bug to enhancement. Just a small observation: in previous versions of tinyb it was working perfectly (and I haven't changed Bluez version since that time) therefore it should be more or less easy to get it working again.

juiice, 2016-10-18:
Hi,

If you fallow the Bluetooth Gatt Services specification, the CCCD (0x2902) and every descriptors in general, belong to a characteristic. It make sense to retrieve a descriptor from the characteristic.
Finally the gatttool command "char-desc" confused me.

mraugit, 2016-10-18:
I've used the maximum setting from this page:
http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide#Movement_Sensor

However, I have seen that this is contrary to the movement settings in the following documentation:
http://www.ti.com/ww/en/wireless_connectivity/sensortag2015/tearDown.html#main

Thanks for the hint.

juiice, 2016-10-18:
When I try to write the CCCD (0x2902) descriptor to enable notification of heart rate sensor, I get this error message:
`tinyb,BluetoothException: GDBus.Error:org.bluezError.NotPermitted: Write not permitted`

When I try to enable the value notification on the heart_rate_measurement (characteristic 0x2A37), I get this error message:
`no class found : tinyb/BluetoothNotification`

When I enable the value notification on the CCCD (0x2902)  descriptor it does not complain, but my callback is never called.

So I try to read the heart_rate_measurement characteristic (0x2A37) value:
`tinyb,BluetoothException: GDBus.Error:org.bluezError.NotPermitted: Read not permitted`.
It is the normal behaviour I think. The characteristic only notify but cannot be read.

I use TinyB commit 5d9cd1f and BlueZ 5.42

I do not know if it is a bug of tinyb, but I cannot make it work. With "gatttool", everything works as is suppose to.
Can you please confirm the only thing I normally have to do, is to enable the value notification on the CCCD (0x2902) and wait for callback.

petreeftime, 2016-10-18:
```no class found : tinyb/BluetoothNotification``` is a TinyB issue, but I'm having some issues replicating it. Are you sure that the ```tinyb.jar``` file you are using is the same that was compiled with the rest of the library?

juiice, 2016-10-18:
It is what I was looking.
Probably my mistake, I re-bundled tinyb to integrate it to my project with Maven.
My apologies.