intel-iot-devkit/tinyb #52: JNI call to findClass BluetoothNotification failed

juiice, 2016-10-20:
Hello,

I am trying to re-bundle "tinyb" for an easier integration into existing project.

The callback of "enable_value_notifications()" cannot find the class 
tinyb/BluetoothNotification.

1 - I build the two C++ libs  "lintinyb.so" and "libjavatinyb.so"
2 - I rebuild the java source and generate a "tinyb-bundle.jar" containing:
- All the java class inside "tinyb" folder
- The 2 natives library "lintinyb.so" and "libjavatinyb.so"
- A LibLoader class to load the native lib

3 - This "tinyb-bundle.jar" is maven dependency of my final application, 
"myapp.jar". When I open "myapp.jar", I can see the "tinyb" folder with every class inside.

At runtime, my "java.class.path" return "myapp.jar".
When I print the list of loaded packages, I can see "tinyb".
No complain java side, everything works. I can load the native libs, get 
my BluetoothAdapter, scan for bt devices, find my device, connect, get 
services ...

But the JNI lib cannot reach java class
Do you have an idea? Where I am wrong?

Attached "tinyb-bundle.jar" re-named as .zip
[tinyb-bundle.zip](https://github.com/intel-iot-devkit/tinyb/files/541579/tinyb-bundle.zip)
 
Regards

juiice, 2016-10-21:
I created a simple java project based on a modified version of "Notification.java" example and my "tinyb-bundle" in dependency.
And it works like a charm. So my "tinyb-bundle" works fine, the problem is somewhere else.
I wonder if it can be a ClassLoader issue cause to multi-threading. What do you think?

petreeftime, 2016-10-21:
I really doubt that it's a multithreading issue. You should not need to call LibLoader, BluetoothManager should load the library when it's instantiated. Do you think you could share the code you are running, or at least the parts of it where you call enableValueNotification?

juiice, 2016-10-21:
I pushed my test further.
Inside my "full project", I tried to start my Bluetooth  communication early (inside the main()), and it works!!!
I have "MyApp" class extending javafx "Application" class. When I start the Bluetooth comm inside the "start()" method of this class, it does not work.

It sound definitively like a ClassLoader issue.

juiice, 2016-10-24:
Hi,

I finally solved my problem. Or at least workaround it.
I cannot find the root cause why the JNI "FindClass" does not work with javafx application. But replacing with JNI "GetObjectClass" works.
```

std::shared_ptr<JNIGlobalRef> callback_ptr(new JNIGlobalRef(callback));

obj_gatt_char->enable_value_notifications([ callback_ptr ] (std::vector<unsigned char> &v)
       {
--            jclass notification = search_class(*jni_env, JAVA_PACKAGE "/BluetoothNotification");
++            jclass notification = jni_env->GetObjectClass(**callback_ptr);
            jmethodID  method = search_method(*jni_env, notification, "run", "(Ljava/lang/Object;)V", false);
            unsigned int size = v.size();

            jbyteArray result = jni_env->NewByteArray((jsize)size);
            jni_env->SetByteArrayRegion(result, 0, (jsize)size, (const jbyte *)&v[0]);

            jni_env->CallVoidMethod(**callback_ptr, method, result);
        });
```

If you are interested by my java packaging, tell me.
For me it is much more convenient to have a "jar" package which embed everything (java classes and natives libraries) instead of installing the natives libs on the system. 
With the LibLoader system, it is easy to support multi-platform.
It is how the "bluecove" multi-platform java bluetooth API works. Nothing to install,  just to add the package dependency in your project.

jonasheinisch, 2016-10-25:
@juiice Could you provide a version for ARM (Raspberry Pi) or could you provide the sources to compile it myself on the RPi?

juiice, 2016-10-25:
@jonasheinisch I do not know if the "libtinyb.so" is architecture dependant.
If not, I can provide you my bundle of course.
If yes, you will have to modify the source of the "tinyb" library. 
petreeftime is the developer of the project. He can probably tell you if the lib work on ARM architecture.

jonasheinisch, 2016-10-25:
@juiice thanks!
@petreeftime can you tell me sth about it? :)

jonasheinisch, 2016-10-26:
@juiice As you already made the changes couldn't you publish a fork of your sources?

juiice, 2016-10-26:
@petreeftime concerning the FindClass issue : strangely I notice the FindClass inside "find()" method of "BluetoothManager" class works fine. So the issue seem to happen only when we try to retrieve "tiny/BluetoothNotification" **interface** from a javafx application. Is it a bug of the javafx ClassLoader?? I do not have the time to dig more deeply. I suggest to use GetObjectClass when retrieving a BluetoothNotification interface. We have a reference of the jobject callback, better using it.

@jonasheinisch I made the change for the FindClass issue. I will publish a fork with a pull request.
concerning the java bundle, now I build the C++ libraries and I copy past them inside a java project. I would like to purpose something cleaner. But I am not a Cmake expert, I progress slowly.

juiice, 2016-10-28:
@jonasheinisch I published my changes.
https://github.com/juiice/tinyb/tree/in-dev
Let me know if it works on ARM. Please create a new issue on my forked project if you have any problem.

petreeftime, 2016-10-28:
The library should work on ARM, I tested it on a RPi some time ago.

The LibLoader idea seems very useful, keep up the good work!

juiice, 2016-10-28:
@petreeftime the LibLoader is ready, it works.
But I am not completely satisfy of the building part. I am not use to cmake. If you can check and clean what I did, it is probably better.
I will push in separate "in-dev" branch for now.

jonasheinisch, 2016-10-28:
@juiice : Thanks, I will try it next week probably and will give an update :)

jonasheinisch, 2016-11-07:
I guess this issue could be closed ;)

closed 2016-11-07 by petreeftime