intel-iot-devkit/tinyb #59: crash if enableConnectedNotifications

gustavo84, 2016-11-25:
Testing the device connection notification causes this error
I have bluez 5.43
#
# JRE version: Java(TM) SE Runtime Environment (8.0_111-b14) (build 1.8.0_111-b14)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.111-b14 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# V  [libjvm.so+0x8de5d6]  Monitor::ILock(Thread*)+0xb6

mjarschel, 2017-01-02:
I can confirm this.

mjarschel, 2017-01-02:
Apparently, this is related to an issue with Bluez itself, since restarting the daemon solved it for me.

adelorenz, 2017-01-04:
Same problem
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007ff8680c45d6, pid=7035, tid=0x00007ff8693d3700
#
# JRE version: Java(TM) SE Runtime Environment (8.0_111-b14) (build 1.8.0_111-b14)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.111-b14 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# V  [libjvm.so+0x8de5d6]  Monitor::ILock(Thread*)+0xb6
```
restarting BlueZ doesn't solve for me

adelorenz, 2017-01-04:
more details:

```
Compilation events (10 events):
Event: 3,653 Thread 0x00007fbe280c1800  292       4       java.util.regex.Pattern$Branch::match (66 bytes)
Event: 3,653 Thread 0x00007fbe280c8800  293       3       java.util.Formatter::locale (9 bytes)
Event: 3,654 Thread 0x00007fbe280c8800 nmethod 293 0x00007fbe191bcf50 code [0x00007fbe191bd0c0, 0x00007fbe191bd348]
Event: 3,661 Thread 0x00007fbe280c1800 nmethod 292 0x00007fbe191c4a50 code [0x00007fbe191c4c60, 0x00007fbe191c5668]
Event: 3,721 Thread 0x00007fbe280c8800  294       3       java.lang.Long::stringSize (38 bytes)
Event: 3,721 Thread 0x00007fbe280c8800 nmethod 294 0x00007fbe191c45d0 code [0x00007fbe191c4740, 0x00007fbe191c4990]
Event: 3,721 Thread 0x00007fbe280c8800  295       3       java.lang.Long::getChars (221 bytes)
Event: 3,722 Thread 0x00007fbe280c8800 nmethod 295 0x00007fbe191bbf50 code [0x00007fbe191bc120, 0x00007fbe191bc770]
Event: 3,992 Thread 0x00007fbe280c8800  296       3       java.util.Formatter$FormatSpecifier::getZero (36 bytes)
Event: 3,993 Thread 0x00007fbe280c8800 nmethod 296 0x00007fbe191c9010 code [0x00007fbe191c91c0, 0x00007fbe191c9708]

GC Heap History (0 events):
No events

Deoptimization events (0 events):
No events

Internal exceptions (7 events):
Event: 0,027 Thread 0x00007fbe2800a800 Exception <a 'java/lang/NoSuchMethodError': Method sun.misc.Unsafe.defineClass(Ljava/lang/String;[BII)Ljava/lang/Class; name or signature does not match> (0x000000076d407c78) thrown at [/HUDSON3/workspace/8-2-build-linux-amd64/jdk8u111/7883/hotspot/src/sh
Event: 0,027 Thread 0x00007fbe2800a800 Exception <a 'java/lang/NoSuchMethodError': Method sun.misc.Unsafe.prefetchRead(Ljava/lang/Object;J)V name or signature does not match> (0x000000076d407f60) thrown at [/HUDSON3/workspace/8-2-build-linux-amd64/jdk8u111/7883/hotspot/src/share/vm/prims/jni.c
Event: 0,106 Thread 0x00007fbe2800a800 Exception <a 'tinyb/BluetoothException': GDBus.Error:org.bluez.Error.InProgress: Operation already in progress> (0x000000076d4dd0a0) thrown at [/HUDSON3/workspace/8-2-build-linux-amd64/jdk8u111/7883/hotspot/src/share/vm/prims/jni.cpp, line 735]
```

mjarschel, 2017-01-09:
I saw that error once more, because I had made a threading error, i.e., asking a thread in Java to sleep when the code was actually run in the native C.