intel-iot-devkit/tinyb #63: Update Intel Edison's tinyB library

yongnan0809, 2016-12-03:
Hello

I wanna use the latest version of tinyB library on edison. 
And I Install the tinyB library 

cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILDJAVA=ON ..
make 
make install

But when I run the Java program have an error

 Native code library failed to load.
java.lang.UnsatisfiedLinkError: /usr/lib/libjavatinyb.so.0.5.0-8-g6e580f4.0.5.0-8-g6e580f4.0.5.0-8-g6e580f4: libawt.so: cannot open shared object file: No such file or directory

any body have idea? 

Thank you.

IOOOTAlan, 2016-12-05:
Hi @yongnan0809 ,
you have to add to your LD_LIBRARY_PATH these paths:
```sh
${JAVA_HOME}/lib/ARCH
${JAVA_HOME}/jre/lib/ARCH
```
where
 - `${JAVA_HOME}` points to your JDK root
 - `ARCH` is your current runtime architecture (amd64, armhf etc.)

For example, on a x86_64 system, you could run your java program with:
```sh
LD_LIBRARY_PATH="${LD_LIBRARY_PATH}:${JAVA_HOME}/lib/amd64:${JAVA_HOME}/jre/lib/amd64" java MainClass
```

yongnan0809, 2016-12-06:
Hi @IOOOTAlan 

Thank you ! 
I solved problem

LD_LIBRARY_PATH="${LD_LIBRARY_PATH}:${JAVA_HOME}/lib/i386:${JAVA_HOME}/jre/lib/i386"
cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DBUILDJAVA=ON
make 
make install 
export LD_LIBRARY_PATH="/usr/java/jre/lib/i386"

And run the java program.

Successfully update the tinyB library on Edison.

Thank you!!

IOOOTAlan, 2016-12-06:
You're welcome ;-)
Please close the issue if it is resolved.

closed 2016-12-06 by yongnan0809