intel-iot-devkit/tinyb #66: Disconnect SensorTag will cause Java Runtime Enviroment fatal error.

yongnan0809, 2016-12-15:
Hello. It's me again.
I Upgared my Edison's tinyB library to newest version of 
https://github.com/intel-iot-devkit/tinyb/tree/6e580f494ced312ed8f56bcd3fc33499d91720e3

And Use Java Sample Notification code with a tiny little modification to test sensor.
Here is my java code 

import tinyb.*;
import java.util.*;
import java.time.*;
import java.util.concurrent.locks.*;
class ValueNotification implements BluetoothNotification<byte[]> {

    public void run(byte[] tempRaw) {
            System.out.print("Temp raw = {");
            for (byte b : tempRaw) {
                System.out.print(String.format("%02x,", b));
            }
            System.out.print("}");
 
            int objectTempRaw = (tempRaw[0] & 0xff) | (tempRaw[1] << 8);
            int ambientTempRaw = (tempRaw[2] & 0xff) | (tempRaw[3] << 8);

            float objectTempCelsius = Notification.convertCelsius(objectTempRaw);
            float ambientTempCelsius = Notification.convertCelsius(ambientTempRaw);

            System.out.println(
                    String.format(" Temp: Object = %fC, Ambient = %fC", objectTempCelsius, ambientTempCelsius));

    }

}

class ConnectedNotification implements BluetoothNotification<Boolean> {
    public void run(Boolean connected) {
            System.out.println("Connected");
    }

}

public class Notification {
    private static final float SCALE_LSB = 0.03125f;
    static boolean running = true;

    static void printDevice(BluetoothDevice device) {
        System.out.print("Address = " + device.getAddress());
        System.out.print(" Name = " + device.getName());
        System.out.print(" Connected = " + device.getConnected());
        System.out.println();
    }

    static float convertCelsius(int raw) {
        return raw / 128f;
    }

    public static void main(String[] args) throws InterruptedException {

        if (args.length < 1) {
            System.err.println("Run with <device_address> argument");
            System.exit(-1);
        }

   
        BluetoothManager manager = BluetoothManager.getBluetoothManager();
        boolean discoveryStarted = manager.startDiscovery();

        System.out.println("The discovery started: " + (discoveryStarted ? "true" : "false"));

        BluetoothDevice sensor = manager.find(null, args[0], null, Duration.ofSeconds(10));

        if (sensor == null) {
            System.err.println("No sensor found with the provided address.");
            System.exit(-1);
        }

        sensor.enableConnectedNotifications(new ConnectedNotification());

        System.out.print("Found device: ");
        printDevice(sensor);

        if (sensor.connect())
            System.out.println("Sensor with the provided address connected");
        else {
            System.out.println("Could not connect device.");
            System.exit(-1);
        }

        //manager.stopDiscovery();

        Lock lock = new ReentrantLock();
        Condition cv = lock.newCondition();

        Runtime.getRuntime().addShutdownHook(new Thread() {
            public void run() {
                running = false;
                lock.lock();
                try {
                    cv.signalAll();
                } finally {
                    lock.unlock();
                }
            }
        });
        BluetoothGattService tempService = sensor.find( "f000aa00-0451-4000-b000-000000000000");

        if (tempService == null) {
            System.err.println("This device does not have the temperature service we are looking for.");
            sensor.disconnect();
            System.exit(-1);
        }
        System.out.println("Found service " + tempService.getUUID());

        BluetoothGattCharacteristic tempValue = tempService.find("f000aa01-0451-4000-b000-000000000000");
        BluetoothGattCharacteristic tempConfig = tempService.find("f000aa02-0451-4000-b000-000000000000");
        BluetoothGattCharacteristic tempPeriod = tempService.find("f000aa03-0451-4000-b000-000000000000");

        if (tempValue == null || tempConfig == null || tempPeriod == null) {
            System.err.println("Could not find the correct characteristics.");
            sensor.disconnect();
            System.exit(-1);
        }

        System.out.println("Found the temperature characteristics");

        byte[] config = { 0x01 };
        tempConfig.writeValue(config);
        
        byte[] period = { 100 };
        tempPeriod.writeValue(period);

        tempValue.enableValueNotifications(new ValueNotification());

       //// ////////////////////////I added stop code here////////////////////////
        while(running){
        	if(!sensor.getConnected())
        		running = false;
        }
        //// ////////////////////////I added stop code here////////////////////////

        lock.lock();
        try {
            while(running)
                cv.await();
        } finally {
            lock.unlock();
        }
        sensor.disconnect();
    }
}

And when sensor poweroff or out of bluetooth signal range program will end with some Error message

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xb7163983, pid=1551, tid=0xb6cf5b40
#
# JRE version: Java(TM) SE Runtime Environment (8.0_101-b13) (build 1.8.0_101-b13)
# Java VM: Java HotSpot(TM) Client VM (25.101-b13 mixed mode linux-x86 )
# Problematic frame:
# V  [libjvm.so+0x46d983]  Monitor::ILock(Thread*)+0xa3
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/root/hs_err_pid1551.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
Aborted

And log file recorded 2 erros.

Internal exceptions (2 events):
Event: 0.194 Thread 0xb6b09400 Exception <a 'java/lang/NoSuchMethodError': Method sun.misc.Unsafe.defineClass(Ljava/lang/String;[BII)Ljava/lang/Class; name or signature does not match> (0xa5806ea0) thrown at [/HUDSON/workspace/8-2-build-linux-i586/jdk8u101/7261/hotspot/src/share/vm/pri
Event: 0.194 Thread 0xb6b09400 Exception <a 'java/lang/NoSuchMethodError': Method sun.misc.Unsafe.prefetchRead(Ljava/lang/Object;J)V name or signature does not match> (0xa5807170) thrown at [/HUDSON/workspace/8-2-build-linux-i586/jdk8u101/7261/hotspot/src/share/vm/prims/jni.cpp, line 4

I'm using Oracle Jdk 1.8.0_101 and 5.43 Version of Bluez on Edison.
So. what should I do ? or how can I terminate my program without fatal error ??
Thank you.

petreeftime, 2016-12-15:
It seems to be a locking problem. I'd try to make the code in the shutdown hook execute only if running is true.

yongnan0809, 2016-12-15:
@petreeftime Hello.
Eh... sorry I didn't  understand your suggestion.
Did you mean let me remove all the lock on my code ? Or change running to true?

I have try to remove all the lock on my code. And when sensor suddenly off the program will show same problem.

Thank you.

petreeftime, 2016-12-15:
I meant changing this piece of code to look like this:
``` 
Runtime.getRuntime().addShutdownHook(new Thread() {
        public void run() {
            if (running) {
                running = false;
                lock.lock();
                try {
                    cv.signalAll();
                } finally {
                    lock.unlock();
               }
            }
        }
    });
```

yongnan0809, 2016-12-17:
ok I will try it.

yongnan0809, 2016-12-20:
@petreeftime  I'm so  Sorry about late reply.

I tried  this code. But it not work.
Same problem shows.

code 
```
Lock lock = new ReentrantLock();
		Condition cv = lock.newCondition();

		Runtime.getRuntime().addShutdownHook(new Thread() {
	        public void run() {
	            if (running) {
	                running = false;
	                lock.lock();
	                try {
	                    cv.signalAll();
	                } finally {
	                    lock.unlock();
	               }
	            }
	        }
	    });

**some codes here ...................**

 while(running){
		 if(!sensor.getConnected())
		    		running = false;
		  }	
	
		lock.lock();
		try {
			while (running)
				cv.await();
		} finally {
			lock.unlock();
		}
		lsmValue.disableValueNotifications();
		skinTempValue.disableValueNotifications();
		ambTempValue.disableValueNotifications();
		sensor.disconnect();
		System.out.println("ByeBye");
```

 Error Message
```
ByeBye
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xb7113983, pid=692, tid=0xb6ca5b40
#
# JRE version: Java(TM) SE Runtime Environment (8.0_101-b13) (build 1.8.0_101-b1                                                                                                                                                             3)
# Java VM: Java HotSpot(TM) Client VM (25.101-b13 mixed mode linux-x86 )
# Problematic frame:
# V  [libjvm.so+0x46d983]  Monitor::ILock(Thread*)+0xa3
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
# An error report file with more information is saved as:
# /home/root/hs_err_pid692.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
Aborted
```
I think it may not cause by lock. 
Is this may cause by jdk problem??

closed 2017-02-03 by yongnan0809