intel-iot-devkit/tinyb #67: [Java] NullPointerException at BluetoothManager.getBluetoothManager()

JafarKhQ, 2016-12-20:
I build the **tinyB** from source and used it in a Java app. (Ubuntu & Raspbian)  

When I run the App from IDE (intelliJ IDEA) it works fine, but after build the jar (Gradle), and run it form terminal, I got a **NullPointerException** At  [BluetoothManager.java line 221](https://github.com/intel-iot-devkit/tinyb/blob/master/java/BluetoothManager.java#L221).

```
public static synchronized BluetoothManager getBluetoothManager() throws RuntimeException, BluetoothException
{
    ..
    .
    String APIVersion = BluetoothManager.class.getPackage().getSpecificationVersion();
    // APIVersion is null !!!
    if (APIVersion.equals(nativeAPIVersion) == false) {
        ...
        ..
        .
    }
}
```

As a temporary workaround I just removed this API version checking code.

petreeftime, 2017-01-07:
Hi, the most likely cause is that the gradle build does not package the MANIFEST file properly since it first needs to be generated through CMake. Someone needs to look into adding Gradle support first for this to work properly.

a333klm, 2020-11-10:
Is there any progress?