intel-iot-devkit/tinyb #68: DeviceManager.getDevices returns not only BLE devices

vkolotov, 2016-12-30:
Hi there, you have been doing a great job guys, keep up!

Just noticed that TinyB actually discovers all devices, including those that do not support BLE (or just do not have any services), e.g. BluetoothManager.getBluetoothManager().getDevices() returns ALL devices in range.

How do I filter out devices that do not support BLE?

The reason why is that when I call "connect" method on a device instance that does not support BLE, it fails with an error (depends on device type).

Thanks,
Vlad

vkolotov, 2016-12-30:
Just noticed that BluetoothDevice.getBluetoothClass returns "0" from BLE enabled devices, and non "0" value for other devices. Is this a right way to identify BLE devices? Thanks.

Be1thaz0r, 2017-03-31:
Hi,
I have the same question...
@vkolotov Did you find the answer on your question?

Thanks in advance