intel-iot-devkit/tinyb #69: Some java methods throw UnsatisfiedLinkError

vkolotov, 2016-12-30:
Calling 
BluetoothDevice.getTxPower 
BluetoothDevice.getBluetoothType

result in 

java.lang.UnsatisfiedLinkError: tinyb.BluetoothDevice.getTxPower()S
java.lang.UnsatisfiedLinkError: tinyb.BluetoothDevice.getBluetoothType()Ltinyb/BluetoothType;

Bluez 5.43 and 0.5.0 tinyb

impala454, 2017-06-12:
`getBluetoothType()` appears to be a virtual method in the C++ side, maybe JNI doesn't like that?  The `getTxPower` method is definitely implemented.  I'll try that function on the C++ side.

vkolotov, 2017-06-12:
great, if you could update c++ stuff, possibly I would be able to fix java part.

lsimedia, 2017-08-10:
Line 1101 in java/jini/BluetoothDevice.cxx
jshort Java_tinyb_BluetoothDevice_getT**X**Power(JNIEnv *env, jobject obj)
should be
jshort Java_tinyb_BluetoothDevice_getT**x**Power(JNIEnv *env, jobject obj)

lsimedia, 2017-08-12:
I submiited a patch to correct the UnsatisfiedLinkError for
BluetoothDevice.getTxPower
BluetoothDevice.getBluetoothType