intel-iot-devkit/tinyb #70: Can't writeValue

mjarschel, 2016-12-30:
Hi,

I'm trying to use tinyb (Java) to control my BLE device (Ubuntu 16.04, Bluez 5.37 + -E flag). Discovering devices, services, and charateristics works. However, when I am trying to use writeValue (like in the sample app) to write to a characteristic, I get the following exception:

tinyb.BluetoothException: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist

Using the bluetoothctl cli-tool, I can send the commands without an issue.

Any idea how to get around this?

petreeftime, 2016-12-30:
You need to update BlueZ to version 5.42 or newer, or use TinyB 0.5.

mjarschel, 2016-12-30:
Upgrading to Bluez 5.43 helped indeed. 
Thanks a lot!

closed 2016-12-30 by mjarschel