intel-iot-devkit/tinyb #75: java.lang.UnsatisfiedLinkError: no javatinyb in java.library.path

vangapandushankar, 2017-01-17:


vivekze, 2017-02-08:
Did you figure out what was wrong here?

vangapandushankar, 2017-02-08:
Hi vivek,

I solved that issue by giving the below command.

 

java -Djava.library.path=javatinyb /usr/local/tinyb/build/examples/java/HelloTinyB.jar:/usr/local/tinyb/build/java/tinyb.jar HelloTinyB 00:A0:50:D0:95:4B

 

thanks,

shankar.v

Be1thaz0r, 2017-02-08:
I had the same problem. On Debian I solved it by modifying LD_LIBRARY_PATH and telling where to look for the library.
Here we are:

echo $LD_LIBRARY_PATH
/usr/local/lib:/usr/lib:/usr/lib/jvm/java-8-openjdk-amd64/jre/lib/amd64

I installed tinyb in the default location (/usr/local/lib). I then added it to LD_LIBRARY_PATH. It happened to me that it then asked for libjawt.so etc. that in my case are located in /usr/lib/jvm/java-8-openjdk-amd64/jre/lib/amd64, that's why you see this line in my example.

After that you can run an example normally as indicated here: [https://software.intel.com/en-us/java-for-bluetooth-le-apps]:

java -cp /usr/local/lib/java/tinyb.jar:examples/java/HelloTinyB.jar HelloTinyB 0E:ED:69:9C:D3:C4

Hope it helps.

AaronBedward, 2017-03-22:
Neither fixes work for me

i have placed the jar in /usr/lib/java and /usr/local/lib

Is there anything else need to do?

I'm using Fedora

IOOOTAlan, 2017-03-23:
@AaronBedward the `java.lang.UnsatisfiedLinkError: no javatinyb in java.library.path` refers to the libjavatinyb.so dynamic library file. If you have this same error, then you just need to put that .so library in a location where the loader will find it (link /usr/lib*) or just override whatever default settings your system has using the `LD_LIBRARY_PATH` environment variable override, just like @Be1thaz0r said.
See also #63 if you have problems using the library.

AaronBedward, 2017-03-23:
@Be1thaz0r Figured it out 

My problem was was that i needed to change the default install path to /usr/ rather than the default /usr/local

settting the `LD_LIBRARY_PATH` didn't work

martinnaughton, 2017-07-31:
its looking for these 2 files

```
libjavatinyb.so
libtinyb.so
```
located what directory they are in then add them in the parameter

In my build directory i ran it like this. The 2 files are located in 2 different directories. This got it working for me.
java -Djava.library.path=/usr/local/lib/:./java/jni/ -cp examples/java/HelloTinyB.jar:./java/tinyb.jar HelloTinyB

maggu2810, 2017-08-30:
> It happened to me that I have been then asked for libjawt.so etc. that in my case are located in /usr/lib/jvm/java-8-openjdk-amd64/jre/lib/amd64

Should be solved by #117

andy-huaan, 2017-09-15:
I had the same problem，My solution as follows：

1. If you run with the Java command，**You need to specify the location of the “libjavatinyb.so” and "libtinyb.so" file to java.library.path**.  for example   `java -Djava.library.path=/usr/local/tinyb/build/java/jni:/usr/local/tinyb/build/src/ -cp examples/java/HelloTinyB.jar:./java/tinyb.jar HelloTinyB`

2.If you run with eclipse or other IDE，You can put the  “libjavatinyb.so” and "libtinyb.so"  file in your project  folder（Can also not be put in）， Then ,`your project-->Properties-->Java Build Path-->Libraries-->JRE SystemLibrary-->Native library location-->Edit`，Select the directory where the file is located.

Hope it helps.

chocomo99, 2022-01-01:
I'm using Eclipse 4.12, JFX11 on Ubuntu 20.04. Built a project that opens a JFX BorderPane with various functions. That works fine. So then I built the tinyb library, created tinyb.jar, added it to the project and on build got the UnsatisfiedLinkError stated above,  Thanks to this thread I added libjavatinyb.so and libtinyb.so to the Native Library for tinyb.jar. It then builds but now crashes with the error below when trying to run the project. Somehow adding the .so files to the tinyb native lib build path causes the new problem with JFX. What now....?

Graphics Device initialization failed for :  es2, sw
Error initializing QuantumRenderer: no suitable pipeline found
java.lang.RuntimeException: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer.getInstance(QuantumRenderer.java:280)
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumToolkit.init(QuantumToolkit.java:222)
	at javafx.graphics/com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:260)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(LauncherImpl.java:658)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:409)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:363)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at java.base/sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:1051)
Caused by: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:94)
	at javafx.graphics/com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	at java.base/java.lang.Thread.run(Thread.java:829)
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at java.base/sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:1051)
Caused by: java.lang.RuntimeException: No toolkit found
	at javafx.graphics/com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:272)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at javafx.graphics/com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.startToolkit(LauncherImpl.java:658)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:409)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:363)
	... 5 more