intel-iot-devkit/tinyb #77: Direct use BLE commands

yongnan0809, 2017-01-29:
Hello. 
Is there have any function or method at tinyB library to directly use ble operating command?
Like these ble command, normaly use BluetoothGattCharacteristic::write_value to send Value 01 to set device.
But now I want directly send specific command to device.
Send all data  " 01 92 FD 05 00 00 32 00 01 "

`-Type           : 0x01 (Command)
-OpCode         : 0xFD92 (GATT_WriteCharValue)
-Data Length    : 0x05 (5) byte(s)
 ConnHandle     : 0x0000 (0)
 Handle         : 0x0032 (50)
 Value          : 01
Dump(Tx):
0000:01 92 FD 05 00 00 32 00 01 `

Thank you.

petreeftime, 2017-01-29:
Hi, you should be able to send an array of bytes to the  device rather than a single value:
```
std::vector<unsigned char> value { 0x01, 0x92, 0xFD, 0x05, 0x00, 0x00, 0x32, 0x00, 0x01 };
characteristic->write_value(value);
```

yongnan0809, 2017-01-30:
@petreeftime Thanks for your reply.
The Value what I need to  send is only 0x01, the other bytes are low level ble commands.
In normal case send some value like you said send byte array use write_value method.
But now I need to send full low level ble commands incluse type , operate code, packet length handle etc.    
Is tinyb library have method or function what I need??

Thank you.

closed 2017-02-03 by yongnan0809