intel-iot-devkit/tinyb #78: Run three program on same time

yongnan0809, 2017-02-03:
Hello. 

I use library run three same program on one Intel edison board, It's gone wrong.
It's perfect run two same program at same time.
but when I execute the program third time it's show some error. 

The program is receive sensor value notification from sensor tag and 1 second receive 50 values.

root@Edison:~/ble# ./SensorTag A0:E6:F8:C1:EC:07
Started = true
Connected CC2650 SensorTag
Waiting for service f000aa80-0451-4000-b000-000000000000to be discovered
terminate called after throwing an instance of 'tinyb::BluetoothException'
what():  GDBus.Error:org.bluez.Error.Failed: No discovery started
Aborted

Is this hardware limit or program error ?

petreeftime, 2017-02-04:
Are you connecting to the same sensor? If it's three separate sensors it should be fine, Edison should be able to connect up to 8 devices I think. If it's the same sensor, I guess it might be some sort of timing error. To be honest, I never thought anyone would want to connect to one sensor from multiple programs at the same time, and it would be pretty difficult to account for this type of behavior. I don't see why it wouldn't work, but you probably need to have a single program connect to the device and then the other programs wait until the first program has finished connecting.

yongnan0809, 2017-02-06:
I'm connect three different sensors(Three sensor tag) at same time(One by one execute program).

Before this I made a program in JAVA. The program is design to receive multiple sensor's data(connect sensor one by one), and made it re-connectable when sensor disconnected by some issues(like long distance caused disconnect, complex signal environment etc.)
And function used at that time is **read_value();**. 
Because this function is request / response type communication, it's performance 10Hz(10 data at 1 sec) speed at best condition.

But the program I need to design is 1 second receive 50 data, and only one way can reach this speed is use notification features receive data form sensors.
Because the tinyb library comes with latest edison system is version 0.4, and notification future is need tinyb 0.5+ &  Bluez 5.4+,so I updated tinyb to last committed tinyb library, update blueZ to 5.43. 
But after update the problem like [Fatal error](https://github.com/intel-iot-devkit/tinyb/issues/66) show up, and less unstable than before. 

In order to avoid  JAVA program's fatal error problem and some unknown issue I changed language to C++ to receive data.

The program source file is [notification](https://github.com/intel-iot-devkit/tinyb/blob/master/examples/notifications.cpp) example file.
Only I changed the source is UUID and sensor settings to receive movement(gyro, accelerator) data at 50Hz(receive 50 data at 1 second).

Program is running by ./SensorTag A0:E6:F8:C1:EC:07 command and start another program until the connection is complete and start receiving data from sensor.
The program is stable and able run two programs(connect two sensors) at same time.
But when running third program(same as first, second program) it's show error.

`terminate called after throwing an instance of 'tinyb::BluetoothException'`
`what(): GDBus.Error:org.bluez.Error.Failed: No discovery started`
`Aborted`

error caused by `ret = manager->stop_discovery();`
After this error, can't run anymore programs and connect any other sensors. Because the error will always show up.
But the program(two programs that already running) is no problem receive data.
And at this state  if sensor disconnect by any reason, it will not able to reconnect, same error show up.
The way to solve this problem is reboot edison board(I didn't find other way to reset bluetooth).