intel-iot-devkit/tinyb #80: SetScanFilter is needed

vkolotov, 2017-02-13:
Hi @petreeftime, @andreivasiliu2211,

thank you for your effort, it is a great lib.

I'm wondering if you could add Adapter.SetDiscoveryFilter method to tinyb library? This is really needed by a plugin for OpenHAB project.

https://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/adapter-api.txt

Particularly we are interested in the "RSSI" filter. 

Thanks,
Vlad

impala454, 2017-06-15:
Hey @vkolotov I implemented this, however have zero time to test it tonight.  If you have the time to test please test the latest master of my fork [here](https://github.com/impala454/tinyb).  Otherwise I can get to it in a day or two.

vkolotov, 2017-06-15:
thanks mate, I'm busy now as well. Will be able to test it on weekends.

However, I'm not quite sure how would I bind std::vector to a java object (array or list)...

petreeftime, 2017-06-15:
You need to declare in java class like so: https://github.com/intel-iot-devkit/tinyb/blob/master/java/BluetoothGattService.java#L91

Then, on the C++ side you need a function like so: https://github.com/intel-iot-devkit/tinyb/blob/master/java/jni/BluetoothGattService.cxx#L145, more specifically, this function: https://github.com/intel-iot-devkit/tinyb/blob/master/java/jni/helper.hpp#L84 converts a C++ vector of type T to a java List of objects of type T.

vkolotov, 2017-09-27:
@petreeftime I finally got down to this. That helper method you mentioned above (https://github.com/intel-iot-devkit/tinyb/blob/master/java/jni/helper.hpp#L84) converts a c++ object to a java object. In our case we need to convert the other way around. Do we have that method already anywhere? Any examples? Thanks.

closed 2017-10-14 by petreeftime