intel-iot-devkit/tinyb #82: Build failing on OSX 10.12

gustavo84, 2017-02-15:
Command "make" ok and finissed correctly.
Command cmake .. -DBUILDJAVA=ON-std=c++11-DCMAKE_C_COMPILER=/usr/bin/clang -DCMAKE_CXX_COMPILER=/usr/bin/clang++ give next error but i don't know the cause of the error:

[  3%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothObject.cpp.o
In file included from /Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:25:
/Users/colinshewell/Downloads/tinyb-master/api/tinyb/BluetoothObject.hpp:80:24: error: implicit
      instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
    static std::string java_class() {
                       ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
In file included from /Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:25:
/Users/colinshewell/Downloads/tinyb-master/api/tinyb/BluetoothObject.hpp:81:16: error: implicit
      instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
        return std::string(JAVA_PACKAGE "/BluetoothObject");
               ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:30:30: error: implicit
      instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
std::string BluetoothObject::get_java_class() const
                             ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:32:11: error: implicit
      instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
   return std::string(JAVA_PACKAGE "/BluetoothObject");
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:35:30: error: implicit
      instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
std::string BluetoothObject::get_class_name() const
                             ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:37:11: error: implicit
      instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
   return std::string("BluetoothObject");
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:40:30: error: implicit
      instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
std::string BluetoothObject::get_object_path() const
                             ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:42:11: error: implicit
      instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits<char>,
      std::__1::allocator<char> >'
   return std::string();
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note: 
      template is declared here
    class _LIBCPP_TYPE_VIS_ONLY basic_string;
                                ^
8 errors generated.
make[2]: *** [src/CMakeFiles/tinyb.dir/BluetoothObject.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/tinyb.dir/all] Error 2
make: *** [all] Error 2

petreeftime, 2017-02-15:
Duplicate of #72

closed 2017-02-15 by petreeftime

gustavo84, 2017-02-16:
thankyou