intel-iot-devkit/tinyb #84: Build failing on macOS Sierra 10.12

gustavo84, 2017-02-16:
Sorry yesterday, I got confused describing the error.

Command cmake .. -DBUILDJAVA=ON -DCMAKE_C_COMPILER=/usr/bin/clang -DCMAKE_CXX_COMPILER=/usr/bin/clang++ finished and run ok:

Command "make" give this error:

[ 3%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothObject.cpp.o
In file included from /Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:25:
/Users/colinshewell/Downloads/tinyb-master/api/tinyb/BluetoothObject.hpp:80:24: error: implicit
instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits,
std::__1::allocator >'
static std::string java_class() {
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note:
template is declared here
class _LIBCPP_TYPE_VIS_ONLY basic_string;
^
In file included from /Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:25:
/Users/colinshewell/Downloads/tinyb-master/api/tinyb/BluetoothObject.hpp:81:16: error: implicit
instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits,
std::__1::allocator >'
return std::string(JAVA_PACKAGE "/BluetoothObject");
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note:
template is declared here
class _LIBCPP_TYPE_VIS_ONLY basic_string;
^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:30:30: error: implicit
instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits,
std::__1::allocator >'
std::string BluetoothObject::get_java_class() const
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note:
template is declared here
class _LIBCPP_TYPE_VIS_ONLY basic_string;
^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:32:11: error: implicit
instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits,
std::__1::allocator >'
return std::string(JAVA_PACKAGE "/BluetoothObject");
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note:
template is declared here
class _LIBCPP_TYPE_VIS_ONLY basic_string;
^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:35:30: error: implicit
instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits,
std::__1::allocator >'
std::string BluetoothObject::get_class_name() const
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note:
template is declared here
class _LIBCPP_TYPE_VIS_ONLY basic_string;
^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:37:11: error: implicit
instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits,
std::__1::allocator >'
return std::string("BluetoothObject");
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note:
template is declared here
class _LIBCPP_TYPE_VIS_ONLY basic_string;
^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:40:30: error: implicit
instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits,
std::__1::allocator >'
std::string BluetoothObject::get_object_path() const
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note:
template is declared here
class _LIBCPP_TYPE_VIS_ONLY basic_string;
^
/Users/colinshewell/Downloads/tinyb-master/src/BluetoothObject.cpp:42:11: error: implicit
instantiation of undefined template 'std::__1::basic_string<char, std::__1::char_traits,
std::__1::allocator >'
return std::string();
^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iosfwd:193:33: note:
template is declared here
class _LIBCPP_TYPE_VIS_ONLY basic_string;
^
8 errors generated.
make[2]: *** [src/CMakeFiles/tinyb.dir/BluetoothObject.cpp.o] Error 1
make[1]: *** [src/CMakeFiles/tinyb.dir/all] Error 2
make: *** [all] Error 2

adamdossa, 2017-03-06:
For anyone following this thread, the following seems to be the solution:

1) In tinyb/api/tinyb/BluetoothObject.hpp add:
#include <string>
#include <sstream>
#include <iostream>

Once I'd done the above, it compiled, but failed with a linking error:
[ 32%] Linking CXX shared library libtinyb.dylib
ld: library not found for -lintl
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/libtinyb.0.5.0-8-g6e580f4.0.5.0-8-g6e580f4.0.5.0-8-g6e580f4.dylib] Error 1
make[1]: *** [src/CMakeFiles/tinyb.dir/all] Error 2
make: *** [all] Error 2

2) Fixing the above required forcing the symlinking of gettext which contains the intl library:
brew install gettext
brew link --force gettext

adamdossa, 2017-03-07:
I've now trying to execute the example code:
java -cp /usr/local/lib/../lib/java/tinyb.jar:examples/java/HelloTinyB.jar -Djava.library.path=/usr/local/lib/ HelloTinyB xxx

But I get the error:
Exception in thread "main" java.lang.RuntimeException: Error getting object manager client: Could not connect: No such file or directory
	at tinyb.BluetoothManager.init(Native Method)
	at tinyb.BluetoothManager.<init>(BluetoothManager.java:207)
	at tinyb.BluetoothManager.getBluetoothManager(BluetoothManager.java:217)
	at HelloTinyB.main(HelloTinyB.java:109)

[I use the MAC address of my BT device above]

Not sure if it is related to needing the BlueZ + GATT stack which is maybe not possible in Mac OSX which has its own stack?

petreeftime, 2017-03-07:
I have already answered here: #83 and #72.
cmake interprets .. as path and will either ignore everything after it or it also interprets it as path. You need to run ```cmake -DBUILDJAVA=ON -std=c++11 -DCMAKE_C_COMPILER=/usr/bin/clang -DCMAKE_CXX_COMPILER=/usr/bin/clang++ ..```, path att the end. However, if you want to run this library on OSX, it won't work since it depends on BlueZ, which is Linux only, and if you want to run on Linux, you will need to cross-compile and that will be quite difficult. Just get a Linux VM.

gustavo84, 2017-03-07:
I thought it was possible to work on mac. I was wrong. 
¿And what alternatives are possible, apart from a Linux virtual machine?

adamdossa, 2017-03-07:
Not sure there are many options - my googling seems to indicate its a linux VM or you're stuck with Obj-C / using Apples stack.

gustavo84, 2017-03-07:
I do not understand how gatt does not have a common implementation for all three operating systems like serial port