intel-iot-devkit/tinyb #86: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method "ReadValue" with signature "a{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist

mikalex, 2017-03-14:
Hello,

I am trying to read data from a wearable sensor (https://www.maximintegrated.com/en/design/reference-design-center/system-board/6147.html) using TinyB. The sensor is discovered, I can connect to it, I can get the services and the characteristics, but when I try to read the value of the characteristics I get the following error:

  ```
Exception in thread "AWT-EventQueue-0" tinyb.BluetoothException: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Method "ReadValue" with signature "a{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
  at tinyb.BluetoothGattCharacteristic.readValue(Native Method)
  at maxrefdes73.TinyBConnectActivity.<init>(TinyBConnectActivity.java:269)
```

I have downloaded and compiled the last version of TinyB and I have Bluez v5.37. I am using Eclipse Neon.

Here is my (simplified) source code (the error is at line: `byte[] raw = characteristic.readValue();` ):

  ```
boolean conn = false;
  try {
  	conn = remoteDevice.connect();
  } catch (BluetoothException be) {
    	System.out.println("Timeout");
  }
        
  if (conn) {
      List<BluetoothGattService> btServices = remoteDevice.getServices();
                       
     for (BluetoothGattService service : btServices) {
    	List<BluetoothGattCharacteristic> btCharacteristics = service.getCharacteristics();
    	              
        for (BluetoothGattCharacteristic characteristic : btCharacteristics) {
             byte[] raw = characteristic.readValue();
        }
    }
 }

```

This is my output:

devicesDiscovered:
 C0:00:00:00:00:00 MAXREFDES73#
Connecting to MAXREFDES73#
CONNECTED to MAXREFDES73#
Services exposed by device and characteristics for each service:

Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
Charateristics:
00002a24-0000-1000-8000-00805f9b34fb
00002a23-0000-1000-8000-00805f9b34fb
00002a29-0000-1000-8000-00805f9b34fb
00002a2a-0000-1000-8000-00805f9b34fb
00002a28-0000-1000-8000-00805f9b34fb
00002a27-0000-1000-8000-00805f9b34fb
00002a26-0000-1000-8000-00805f9b34fb
00002a25-0000-1000-8000-00805f9b34fb

Service UUID: 0000180d-0000-1000-8000-00805f9b34fb
Charateristics:
00002a39-0000-1000-8000-00805f9b34fb
00002a38-0000-1000-8000-00805f9b34fb
00002a37-0000-1000-8000-00805f9b34fb

Service UUID: 00001801-0000-1000-8000-00805f9b34fb
Charateristics:
00002a05-0000-1000-8000-00805f9b34fb

Service UUID: 0000180f-0000-1000-8000-00805f9b34fb
Charateristics:
00002a19-0000-1000-8000-00805f9b34fb

```
$ bluetoothd -v
5.37

$ uname -a
Linux plimbariciSon 4.8.0-39-generic #42~16.04.1-Ubuntu SMP Mon Feb 20 15:06:07 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial
```

I've seen in [this thread](https://github.com/intel-iot-devkit/tinyb/issues/30) that BlueZ 5.43 has introduced parameters for its readValue() method, but I have BlueZ 5.37 so I can't figure out where the problem is. 

Thank you.

petreeftime, 2017-03-14:
There's no backwards compatibility. If you want to use the latest version of TinyB you'll need BlueZ 5.42 or later, otherwise use version 0.5.

mikalex, 2017-03-14:
Thank you for you quick reply.

Isn't 0.5 the latest version of TinyB?

I had initially installed Bluez 5.43 and then reinstalled 5.37 because of the argument taken by readValue() in 5.43.

If BlueZ 5.43 has a method `array{byte} ReadValue(dict options)` , shouldn't there be a `readValue()` method with arguments in TinyB? 
There is `public native byte[] readValue();` in `BluetoothGattCharacteristic.java` and
```
jbyteArray Java_tinyb_BluetoothGattCharacteristic_readValue(JNIEnv *env, jobject obj) {
    try {
        BluetoothGattCharacteristic *obj_gatt_char =
                                    getInstance<BluetoothGattCharacteristic>(env, obj);
        std::vector<unsigned char> array = obj_gatt_char->read_value();
    ...
}
```
in `BluetoothGattCharacteristic.cxx` 

I'm sorry if my questions sound stupid, I'm not very skilled with jni.

mikalex, 2017-03-22:
Apparently the first time I had installed Bluez 5.43 I was actually still running 5.37 (because it was installed in /usr/local/libexec/bluetooth/bluetoothd but the version running was /usr/sbin/bluetoothd). Now I managed to reinstall and actually run Bluez 5.43 (`bluetoothd -v -> 5.43`) (by symlink /usr/sbin/bluetoothd to /usr/local/libexec/bluetooth/bluetoothd: `sudo ln -s /usr/local/libexec/bluetooth/bluetoothd /usr/sbin/bluetoothd`) and I can read the values of the characteristics :)

Thank you and sorry for bothering.

closed 2017-06-10 by petreeftime