intel-iot-devkit/tinyb #87: How to do pairing in TinyB?

yongnan0809, 2017-03-16:
We are working on a project to connect our IoT gateway on the Intel Edison board to a few wearable devices via BLE.
We are currently using the TinyB for the BLE communication.
TinyB is working nicely with most of them.

However, we have found that one of the current wearable devices  requires pairing for BLE communication.
Would you please tell us how to do pairing in TinyB ?
We would appreciate your help very much.

Thank you very much

petreeftime, 2017-06-10:
Just calling the device's pair method should work in most cases. There should be a default handler set up in the system. There's no support for writing your own handler at this point.

rajithavk, 2018-02-20:
Any update on this?
We are also interested to see a mechanism to pair using a pairing key included within this library since most of the practical applications are handled that way.

Really appreciate if someone can point in the direction to develop the feature so that we can give a try at it.

Cheers.