intel-iot-devkit/tinyb #89: List of services

Be1thaz0r, 2017-03-17:
Hi,
I'm trying to get the list of services on my BLE device and observe a weird thing.
When I get the services normally through gatttool I obtain this result:
attr handle: 0x0001, end grp handle: 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x0008, end grp handle: 0x0008 uuid: 0000fff0-0000-1000-8000-00805f9b34fb
attr handle: 0x0009, end grp handle: 0x000b uuid: 0000ffe5-0000-1000-8000-00805f9b34fb
attr handle: 0x000c, end grp handle: 0xffff uuid: 0000ffe0-0000-1000-8000-00805f9b34fb

However, when I get them from tinyb I get this:
Services list size: 3
UUID: 0000ffe0-0000-1000-8000-00805f9b34fb
UUID: 0000ffe5-0000-1000-8000-00805f9b34fb
UUID: 0000fff0-0000-1000-8000-00805f9b34fb

As you can see, I see 4 services when using gatttool and only 3 through tinyb. Could you please help me to understand what's going on?
Thanks in advance.

petreeftime, 2017-03-17:
The first one is the Generic Access Profile Service and it's managed by BlueZ, so it should not be visible to users directly, but through some of BlueZ's interfaces.

Be1thaz0r, 2017-03-17:
Thanks for your reply!