intel-iot-devkit/tinyb #91: RSSI notifications never get fired once the BLE device is connected

tanvn, 2017-03-21:
I want to get the value of the BLE device for every 3 seconds and I found out that when my BLE device is not connected, the RSSI is updated and I can get the value via notification mode. But once I connect my raspberry pi (using bluez) to the BLE device, the RSSI notification is never fired again, and using getRSSI() will return constant value.
Is this a bug of the library or bluez or it's just the specification ?

A reply would be greatly appreciated.

jcfrk101, 2017-05-15:
I have the a similar issue, except I get nothing from notifications once registered. I'm running on 5.0.

petreeftime, 2017-06-10:
This is likely an issue with BlueZ. You should try to open a ticket with them. http://www.bluez.org/contact/