intel-iot-devkit/tinyb #95: BluetoothGattCharacteristic.enableValueNotifications() fails with "Operation is not supported"

swilson, 2017-04-26:
I'm trying to use tinyb on Ubuntu, and BluetoothGattCharacteristic.enableValueNotifications() always fails with:

tinyb.BluetoothException: GDBus.Error:org.bluez.Error.NotSupported: Operation is not supported

This is presumably coming from chr_start_notify() in bluez/tools/gatt-service.c. I can't seem to find a version of bluez that actually supports StartNotify(); is there one? If not, how does the Notification.java example work?

petreeftime, 2017-04-26:
It's supported in BlueZ 5.42 or newer for sure, since it worked when I tested it.

swilson, 2017-04-27:
What platform was that on? Perhaps it is using a fork of BlueZ? It appears the main version (https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/tools/gatt-service.c) doesn't have support for this.

swilson, 2017-04-27:
Had an issue with my service table; tinyb works fine.

closed 2017-04-27 by swilson