intel-iot-devkit/tinyb #96: GetServices return zero in Raspian

flipper83, 2017-05-02:
Hi,

I compiled the version for a Raspian. 
The server find ok the devices correctly, but when I invoke getServices the list is empty.
I've checked and I included:
- the -E in the bluetooth.service and I've checked that is working after restart with the ps
- I've inluded in /etc/dbus-1/system.d/bluetooth.conf under <policy context="default">:
        <allow send_interface="org.bluez.GattService1"/>
        <allow send_interface="org.bluez.GattCharacteristic1"/>
        <allow send_interface="org.bluez.GattDescriptor1"/>
  and I included in the root policy too.

If something missing? :S Can you give me a clue? 

Regards,

petreeftime, 2017-05-02:
Did you restart the device after setting the bluetooth.conf policy?

Did connection to the device succeed? device.getConnected() should return true.

flipper83, 2017-05-04:
yep I solved.
The problem is rasbian comes with an old bluez version.

 I needed to compile new one and overrode the previous one.

Thank you and sorry for not reply until now.

closed 2017-08-14 by petreeftime