intel-iot-devkit/tinyb #98: All ble devices are not detected and scanning is very slow.

ralf601, 2017-05-21:
I am not able to detect all ble devices, there are 150+ beacons in room but i am able to detect only 60~70 beacons and it takes about 1 minute to detect.

On other hand i am able to detect 150+ in approx 1 minute on Samsung galaxy s4.

Here is my code:
```
		BluetoothManager manager = BluetoothManager.getBluetoothManager();
		boolean discoveryStarted = manager.startDiscovery();
		Map<String,BluetoothDevice> devices = new HashMap<>();
		print("The discovery started: " + (discoveryStarted ? "true" : "false"));
		new Thread(new Runnable() {
			@Override
			public void run() {
				// TODO Auto-generated method stub
				while(true){
					for(BluetoothDevice device : manager.getDevices()){
						devices.put(device.getAddress(), device);
					}
					print("Devices Found:"+devices.size());
					try {
						Thread.sleep(1000);
					} catch (InterruptedException e) {
						// TODO Auto-generated catch block
						e.printStackTrace();
					}
				}}
		}).start();
```

petreeftime, 2017-05-21:
This is likely an issue with BlueZ. The only thing TinyB is doing in this case is reading data from it. You should try to open a ticket with them. http://www.bluez.org/contact/

vkolotov, 2017-09-21:
Same here @petreeftime @ralf601 . Not all devices get discovered. However there is workaround. 
Use bluetoothctl tool to set:
```sh
[bluetooth]# set-scan-filter-rssi -100
```
After setting rssi filter to -100, all devices get discovered.

vkolotov, 2017-09-21:
Thats why I created this thicket: https://github.com/intel-iot-devkit/tinyb/issues/80