intel-iot-devkit/tinyb #99: Unable to get Manufacturer Data

ralf601, 2017-05-21:
I am unable to get 

Here is y code:
```
for(BluetoothDevice device : manager.getDevices()){
						Map<Short, byte[]> a = device.getManufacturerData();
						print("ManufacturerData Size:"+a.size());
						for(Short key : a.keySet()){
							print("Key="+key+" Valaue"+Arrays.toString(a.get(key)));	
						}

						devices.put(device.getAddress(), device);
					}
```
Output:

```
ManufacturerData Size:0
ManufacturerData Size:0
ManufacturerData Size:0
ManufacturerData Size:0
ManufacturerData Size:0
ManufacturerData Size:0
```

framspott94, 2017-07-26:
Hi ralf601,

I have the same problem so far... is there any progress you could show?
I successfully connected my Raspi to an BLE Beacon (BeaconInside 2nd Gen), but this does not seem to work...

I would be very grateful, if we could help each other through this problem... ;)

All the best

framspott94

impala454, 2017-07-26:
I have noticed in practice that sometimes what I expect to see as manufacturer data shows up as service data.  Have you tried looking at the service data for your device?

framspott94, 2017-08-01:
Hi impala454,

thanks for the fast reply! I finally could try it now with the service data, but unfortunately it is the same problem, I only get empty maps back... Here is my java code, maybe there is a problem with it...

```
Map<String, byte[]> servicedata = null;
System.out.println("Trying to get service data...");
servicedata = sensor.getServiceData();
        
if(servicedata == null){
       System.out.println("Service data is null...");
       System.exit(0);
}
if(servicedata.isEmpty()){
       System.out.println("Service data is empty...");
       System.out.println(servicedata);
}
```
Thanks for the fast help!

All the best

framspott94

framspott94, 2017-08-01:
Hi again impala454,

additionally I have to say, that I have successfully received Advertisement packets with an SDR. Should I post a Wireshark screenshot of the paket?

All the best

framspott94

framspott94, 2017-08-02:
Today I tried if it is a timing issue, but with a two second break before and after the functions (getManufacturerdata and getServiceData) it also doesnt changes the result.

All the best

framspott94

impala454, 2017-08-02:
Have you tried looping and continuously calling it?  I know some of my devices take a couple of reads before I see the data.

framspott94, 2017-08-02:
Yeah, I tried that for both functions... have you got a java code for that? I also saw yesterday that the manufacturer notifier was gone in the docu.. quite suprising.. ;) have you got pushed a new version since about two weeks?

impala454, 2017-08-02:
I don't do a lot of java but I assume you should be able to just wrap that code with a

    while (true) {
    
    }

I'm not sure what you mean about the missing manufacturer notifier.  If you look at the [master branch commits page](https://github.com/intel-iot-devkit/tinyb/commits/master) you'll see only three small commits since October of last year.

framspott94, 2017-08-02:
No I mean the doxygen documentation... So you were able to do it with c++? I tried it with a while loop, that was one of the first things I tried... ;) Are you a developer of the library?

framspott94, 2017-08-02:
Or is there a way to speak directly to bluez, and to avoid those functions...?

framspott94, 2017-08-03:
I looked at my wireshark capture, and I could see, that my desired information is brougth by service data advertising packets, so you were right. I also could receive it with my raspi by using the console program bluetoothctl ("bluetoothctl" -> "scan on").

impala454, 2017-08-03:
- I am able to use the C++ API to read both manufacturer and service data.  

- I'm  not really a developer of the library but am contributing some and will be a lot more in the near future.

- You probably don't want to attempt to "speak" directly to bluez.  That's what this library does for you.  Bluez uses dbus on the back end to do all its communication and tinyb is an API to those dbus calls. 

- Have you tried using the `enableServiceDataNotifications()` functionality?  I'm not sure if the data you're reading is supposed to be one time or constantly updating but that function (on the C++ side) and the manfuacturer data function works great for me for several devices.  When you're wiresharking the data, are you controlling the device with your phone or some other device?  Often, these devices require some characteristic to be written with some special value before they'll begin updating or notifying properly.

framspott94, 2017-08-03:
Thanks for the answer, I want to attempt it with Java. And I wasnt aware of the function enableServiceDataNotifications()... I already heard of the difficulty of bluez, but hoped it could be a "last option"... The service data is kept in a ADV_IND advertising packet (undirected connectable mode). I will review the characteristics of my device, maybe there is the issue. Thanks for the hints and the help! An update is coming soon... ;)

All the best from austria!

framspott94

framspott94, 2017-08-03:
One thing I forgot is that I get an Segmentation Fault if I implement one of those Notifications like shown in the examples...

impala454, 2017-08-03:
If you could post a small, complete example it might be easier to figure out what's wrong.

framspott94, 2017-08-03:
So there are snippets of my code:

`class AdvertisementNotification implements BluetoothNotification<Map<String, byte[]>> {

    @Override
    public void run(Map<String, byte[]> value) {
        System.out.println("Service Data is recognized!");
        System.out.println("Advertisement Notification received!");
        //throw new UnsupportedOperationException("Not supported yet."); //To change body of generated methods, choose Tools | Templates.
    }

}`

`
        BluetoothManager manager = BluetoothManager.getBluetoothManager();

        boolean discoveryStarted = manager.startDiscovery();

        System.out.println("The discovery started: " + (discoveryStarted ? "true" : "false"));
        BluetoothDevice sensor = getDevice(args[0]);

        try {
            manager.stopDiscovery();
        } catch (BluetoothException e) {
            System.err.println("Discovery could not be stopped.");
        }

        if (sensor == null) {
            System.err.println("No sensor found with the provided address.");
            System.exit(-1);
        }

        System.out.print("Found device: ");
        printDevice(sensor);

sensor.enableServiceDataNotifications(new AdvertisementNotification());
`

There is the Segmentation fault output:

`The discovery started: true
Address = A0:E6:F8:77:AB:0D
 Name = Eingang1poller
 Connected = false
Address = 2C:60:0C:70:93:69 Name = y44ju58x7f
 Connected = false
Discovery could not be stopped.
Found device: 
Address = A0:E6:F8:77:AB:0D
 Name = Eingang1poller Connected = false
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV
 (0xb) at pc=0x767595ec, pid=2371, tid=1982911584
#
# JRE version: Java(TM) SE Runtime Environment (8.0_65-b17) (build 1.8.0_65-b17)
# Java VM: Java HotSpot(TM) Client VM (25.65-b01 mixed mode linux-arm )
# Problematic frame:
# 
V  [libjvm.so+0x3d05ec]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# 
/home/pi/NetBeansProjects/TinyBTest/hs_err_pid2371.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#`

I do also get Segmentation Faults with the Manufacturer Data Notification. Thanks for the help in anticipation!

framspott94, 2017-08-03:
Sorry for the crappy format...

framspott94, 2017-08-03:
So, I looked again at the specification of the beacon, and tried a bit... There is the funny case that I now get continously Service data with bluetoothctl, if my mobile phone is on and a specific app with my bluetooth is activated/turned on. Now I also get sth. with the function getServiceData(). But there is still a problem:

`Trying to get service data...
(process:3446): GLib-CRITICAL **: the GVariant format string 'a{sv}' has a type of 'a{sv}' but the given value has a type of 'a{qv}'
(process:3446): GLib-CRITICAL **: g_variant_get: assertion 'valid_format_string (format_string, TRUE, value)' failed
**
GLib:ERROR:/build/glib2.0-tTvduh/glib2.0-2.42.1/./glib/gvarianttypeinfo.c:170:g_variant_type_info_check: assertion failed (container->ref_count > 0): (-390264832 > 0)`

framspott94, 2017-08-03:
Quiet strange, that fact (with my mobile phone) is nowhere documented... ;)

petreeftime, 2017-08-03:
What version of BlueZ are you using?

petreeftime, 2017-08-03:
I didn't and still don't have a way to test this, but there are two issues with manufacturer data/service data from what I can see: the type of the key is wrong and service data actually gets manufacturer data.

petreeftime, 2017-08-03:
https://github.com/intel-iot-devkit/tinyb/commit/fcf01dbe65938547c453db9d06d859aa7975f4bb should fix your crash, but I don't know why service/manufacturer data would be empty.

framspott94, 2017-08-04:
Hi petreeftime,

I am using Bluez 5.23. Thanks for the fix, I will test it as soon as I can. I think, I now can receive something, I think, because the function crashes, due to a type error. We will see, as soon as I know more I will let you know! Big thanks for the help!

All the best!

framspott94

petreeftime, 2017-08-04:
You need BlueZ 5.42+ for getServiceData/getManufacturerData.

framspott94, 2017-08-04:
Ah, ok, Good tip!

framspott94, 2017-08-04:
Thanks petreeftime, now the function getManufacturerData() gives back an Map! I have now Bluez 5.42 installed and the latest version of tinyb. But the function getServicedata() still gives back the same error, and I need the information, which is coming with the servicedata, I am afraid... Error:

(process:1735): GLib-CRITICAL **: the GVariant format string 'a{sv}' has a type of 'a{sv}' but the given value has a type of 'a{qv}'
(process:1735): GLib-CRITICAL **: g_variant_get: assertion 'valid_format_string (format_string, TRUE, value)' failed
**
GLib:ERROR:/build/glib2.0-tTvduh/glib2.0-2.42.1/./glib/gvarianttypeinfo.c:170:g_variant_type_info_check: assertion failed (container->ref_count > 0): (-390264832 > 0)

But we are finally getting somewhere... ;)

framspott94, 2017-08-07:
Hi again,

could it help, if I reinstall everything on my pi? Maybe there are problems with my previous version of tinyb...? Do I also have to exchange the jar files I added to my netbeans java project?

All the best

framspott94

petreeftime, 2017-08-07:
I don't know if you need to do a full reinstall, but it might help. Patch https://github.com/intel-iot-devkit/tinyb/commit/fcf01dbe65938547c453db9d06d859aa7975f4bb should fix the issue. I tested this via Beacon Toy and it seems to show good data. I'll add a patch to the list_mfg example.

```
Class = BluetoothDevice Path = /org/bluez/hci0/dev_5C_F5_BA_A4_66_6B Name = ASUS_Z00ED Connected = 0
Service
        0000feaa-0000-1000-8000-00805f9b34fb = [ 16, 230, 2, 99, 102, 46, 112, 104, 121, 115, 105, 99, 97, 108, 45, 119, 101, 98, 8, ]
```

petreeftime, 2017-08-07:
This is the updated example: https://github.com/intel-iot-devkit/tinyb/commit/0cb6904ad9ea1846176598dfca7afd1aeb42a50f

framspott94, 2017-08-08:
Thanks for the updated example! Is there a working java version of it? I will do a full reinstall and will report if it helps...

All the best

framspott94

petreeftime, 2017-08-08:
Did you run cmake with -DBUILDJAVA=ON?

You should be able to find the files by using ```find / -iname libjavatinyb.so```.

framspott94, 2017-08-10:
Thanks, no I didnt, last time it worked without that... I'll try it soon!

All the best

framspott94

framspott94, 2017-08-10:
I finally get return values with getserviceData(). Now I will test, if the values are right.

@petreeftime Should I, if it works, send you the testprogram for other folks, who are facing the same issues?

All the best

framspott94

framspott94, 2017-08-10:
I finally successfully managed to read my service data! Big thanks to all! Now follows my little test java code, maybe it helps... ;) :

`public static void main(String[] args) throws InterruptedException {
        if (args.length < 1) {
            System.err.println("Run with <device_address> argument");
            System.exit(-1);
        }
        System.out.println("There are input arguments!");
        System.out.println(args[0]);

        BluetoothManager manager = BluetoothManager.getBluetoothManager();

        boolean discoveryStarted = manager.startDiscovery();

        System.out.println("The discovery started: " + (discoveryStarted ? "true" : "false"));
        BluetoothDevice sensor = getDevice(args[0]);

        try {
            manager.stopDiscovery();
        } catch (BluetoothException e) {
            System.err.println("Discovery could not be stopped.");
        }

        if (sensor == null) {
            System.err.println("No sensor found with the provided address.");
            System.exit(-1);
        }

        System.out.print("Found device: ");
        printDevice(sensor);
        
        while(true){
            Map<String, byte[]> servicedata = null;
            System.out.println("Trying to get service data...");
            //System.out.println(sensor.getName());

            Thread.sleep(2000);

            servicedata = sensor.getServiceData();

            Thread.sleep(2000);

            if(servicedata == null){
                System.out.println("Service data is null...");
                continue;
                //System.exit(0);
            }
            if(servicedata.isEmpty()){
                System.out.println("Service data is empty...");
                System.out.println(servicedata);
                continue;
            }
            System.out.println(servicedata.size());
            System.out.println(servicedata.values());
            
            for(String key : servicedata.keySet()){
		System.out.println("Key=" + key + " Value=" + Arrays.toString(servicedata.get(key)));
                byte[] IDRaw = null;
                try{
                    IDRaw = servicedata.get(key);
                }catch(BluetoothException e){
                    System.out.println("BluetoothException while reading Gatt characteristic");
                }
                if(IDRaw == null){
                    System.out.println("Error while reading Gatt characteristic => null");
                }
                String IDString = "";
                for (byte b : IDRaw) {
                    IDString = IDString + String.format("%02x", b );
                }
                System.out.print("ID raw data = " + IDString);
            }
            
            break;

        }
}`

petreeftime, 2017-08-12:
You could send me a pull request with that example.

I will close the issue now if everything is working.

closed 2017-08-12 by petreeftime

Framspott, 2017-08-12:
Thanks for the proposal! I would be glad if I can contribute! ;)

Receiving the data works fine, but I only get the latest data of bluez, and I am not sure if how old the data is I am receiving. Because I am getting service data even if my beacon device is not powered, although my device sends advertising packets every 100 ms! Therefore I thought, after I received the data, I could delete the device and the data like with the bluetoothctl command "remove <MAC-address>". Then bluez would have to reload the data. Or is there another way to guarantee the that the very latest packet is exposed to tinyb? Also the received RSSI with getRSSI() is wrong...

All the best - I will look for the pull request monday!

framspott94