intel-iot-devkit/tinyb #100: building fails with gcc7

plusky, 2017-06-08:
In current openSUSE Factory, with gcc7 build fails with following:
```
[   48s] cd /home/abuild/rpmbuild/BUILD/tinyb-0.5.0/build/src && /usr/bin/c++  -Dtinyb_EXPORTS -I/home/abuild/rpmbuild/BUILD/tinyb-0.5.0/api -I/home/abuild/rpmbuild/BUILD/tinyb-0.5.0/api/tinyb -I/home/abuild/rpmbuild/BUILD/tinyb-0.5.0/include -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/gio-unix-2.0  -fmessage-length=0 -grecord-gcc-switches -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -DNDEBUG -O2 -g -DNDEBUG -fPIC   -std=gnu++11 -o CMakeFiles/tinyb.dir/BluetoothGattDescriptor.cpp.o -c /home/abuild/rpmbuild/BUILD/tinyb-0.5.0/src/BluetoothGattDescriptor.cpp
[   49s] In file included from /home/abuild/rpmbuild/BUILD/tinyb-0.5.0/src/BluetoothGattDescriptor.cpp:28:0:
[   49s] /home/abuild/rpmbuild/BUILD/tinyb-0.5.0/api/tinyb/BluetoothGattDescriptor.hpp:60:10: error: 'function' in namespace 'std' does not name a template type
[   49s]      std::function<void(std::vector<unsigned char> &)> value_changed_callback;
[   49s]           ^~~~~~~~
[   49s] /home/abuild/rpmbuild/BUILD/tinyb-0.5.0/api/tinyb/BluetoothGattDescriptor.hpp:107:14: error: 'std::function' has not been declared
...
```

petreeftime, 2017-06-10:
Seems I forgot to include the functional header.

panovvv, 2017-07-30:
I made a pull request with the fix: #105

petreeftime, 2017-07-30:
Should be fixed. Thanks @panovvv.

closed 2017-07-30 by petreeftime