intel-iot-devkit/tinyb #103: No Services Found

lhwa561, 2017-07-11:
Hello,

I am currently working with a tinyBLE and I am having problems getting services to access the uuid.
Currently I am on Version 5.44 of BlueZ with the latest raspbian on a raspberry pi 3 model b
I have looked through the other issues and have tried:

-  adding -E in the bluetooth.service in /lib/systemd/system/bluetooth.service
-  added allow send_interface="org.bluez.GattService1" 
allow send_interface="org.bluez.GattCharacteristic1" 
allow send_interface="org.bluez.GattDescriptor1" 
under policy context="default"
-  allow send_interface="org.bluez.GattCharacteristic1"
 allow send_interface="org.bluez.GattDescriptor1" 
are both in policy user="root" as well
-  confirmed that daemon is working
-  confirmed that the kernel supports bluetooth
-  restarted the raspberry pi multiple times

I can connect to the tinyBLE using gatttool and read from the handle in console and when I check device info on bluetoothctl I can see UUID: Vendor Specific, UUID: Generic Access Profile, UUID: Generic Attribute Profile and UUID: Nordic UART Service. 

I am able to connect to the tinyBLE using code from the hellotinyb.cpp 
```
list_devices = manager->get_devices();
while(1) {    
    for (auto it = list_devices.begin(); it != list_devices.end(); it++) {
         if ((*it)->get_address == THE_ADDRESS) {
            sensor_tag = (*it).release();
        }
    }
    if (sensor_tag != nullptr) {
        break;
    }
    else {
        this_thread::sleep_for(chrono::seconds(4));
    }
}

sensor_tag->connect();
```
However when I use
```
while(1) {
    this_thread::sleep_for(chrono::seconds(4));
    auto list = sensor_tag->get_services();
    if (list.empty()) {
        cout << "No services" << endl;
    }
    else {
        cout << "Service Found" << endl;
        return 0;
    }
}
```
I am getting an empty list every time. 

Could I get some help regarding this?

Thanks,

lhwa561, 2017-07-11:
Also I am able to use the nordic uart service and read the tx characteristics with the nrf connect mobile application

impala454, 2017-07-11:
Could you post the full source code?

lhwa561, 2017-07-12:
```
#include <string>
#include <string.h>
#include <cstdlib>
#include <iostream>
#include <stdio.h>
#include <iomanip>
#include <stdint.h>
#include <tinyb.hpp>

#include <vector>
#include <thread>
#include <atomic>
#include <csignal>

using namespace tinyb;
using namespace std;

std::atomic<bool> running(true);

void signal_handler(int signum)
{
    if (signum == SIGINT) {
        running = false;
    }
}

vector<string> list_of_tiny;

bool check_address(string s) {
    bool output = false;

    for (int i = 0; i < list_of_tiny.size(); i++) {
        if (s == list_of_tiny[i]) {
            return true;
        }
    } 	
    return output;
}

int main(int argc, char **argv) 
{
    BluetoothManager *manager = nullptr;
    try {
        manager = BluetoothManager::get_bluetooth_manager();
    } catch(const std::runtime_error& e) {
        cerr << "Error while initializing libtinyb: " << e.what() << endl;
        exit(1);
    }

    cout << "Initialization was succesful." << endl;
	
    /* Start the discovery of devices */
    bool ret = manager->start_discovery();
    if (!ret) {
        cout << "Failed to start Discovery" << endl;
        return 0;
    }
    cout << "Discovering BLE devices" << endl;
    
    std::unique_ptr<BluetoothAdapter> adapter = manager->get_default_adapter();
    BluetoothDevice *ess_device = NULL;
    BluetoothDevice *sensor_tag = NULL;
    BluetoothGattService *tinyb_service = NULL;

    list_of_tiny.push_back("E2:48:FF:ED:4E:3D"); 
    list_of_tiny.push_back("F1:98:00:E3:2D:9C");

    vector<unique_ptr<BluetoothDevice>> list_devices;
    //for (int i = 0; i < 15; i++) {
    while(1) {
//    	cout << "Discovered Devices: " << endl;
        list_devices = manager->get_devices();
	
        for (auto it = list_devices.begin(); it != list_devices.end(); it++) {
            if ((*it)->get_rssi() != 0) {
	    if (check_address((*it)->get_address())) {
                cout << "RSSI = " << (*it)->get_rssi() << " ";
//              cout << "Class = " << (*it)->get_class_name() << " ";
                cout << "Path = " << (*it)->get_object_path() << " ";
//              cout << "Name = " << (*it)->get_name() << " ";
//              cout << "Connected = " << (*it)->get_connected() << " ";
                cout << endl;
            }
            if ((*it)->get_address() == "E2:48:FF:ED:4E:3D") {
                cout << "E2:48:FF:ED:4E:3D address found" << endl;
                sensor_tag = (*it).release();
//              sensor_tag->connect();
            }
            /*
            if ((*it)->get_address() == "F1:98:00:E3:2D:9C") {
            cout << "address found" << endl;
            sensor_tag = (*it).release();
            sensor_tag->connect();
            }	*/
        }
    }
	
    if (sensor_tag != nullptr) {
        break;
    }
    else {
        std::this_thread::sleep_for(std::chrono::seconds(4));
        //cout << endl;
        }
    }
    cout << "Stopping Discovery... " << endl;
    ret = manager->stop_discovery();
    cout << "Discovery Stopped" << endl;	
    if (sensor_tag == nullptr) {
    	cout << "Could not find device " << "E2:48:FF:ED:4E:3D" << endl;
    	return 0; 
    }
    sensor_tag->connect();

    /* Wait for the device to come online */
    cout << "Device Coming Online... " << endl;
    std::this_thread::sleep_for(std::chrono::seconds(4));
    cout << "Obtaining Sensor Tag services" << endl; 
    while (1) {		
    	std::this_thread::sleep_for(std::chrono::seconds(4));
        auto list = sensor_tag->get_services();
        if (list.empty()) {
            cout << "No services Found" << endl;
			 
            //continue;
        }
        else {
             cout << "Services Found" << endl;
            /*
            for (auto it = list.begin(); it != list.end(); ++it) {
                std::cout << "Class = " << (*it)->get_class_name() << " ";
                std::cout << "Path = " << (*it)->get_object_path() << " ";
                std::cout << "UUID = " << (*it)->get_uuid() << " ";
                std::cout << "Device = " << (*it)->get_device().get_object_path() << " ";
                std::cout << std::endl;
                // Search for the tinyb service, by UUID 
                if ((*it)->get_uuid() == "") {
                tinyb_service = (*it).release();
                }
            }
            */
            return 0;
        }
    }
    return 0;
}

```

impala454, 2017-07-12:
You mentioned you're able to read the device with gatttool, have you tried directly reading it via UUID from the code (skipping the `get_services()`) call?

lhwa561, 2017-07-12:
How would I go about doing that?

lhwa561, 2017-07-12:
nvm found it

impala454, 2017-07-12:
[This example](https://github.com/intel-iot-devkit/tinyb/blob/master/examples/asynctinyb.cpp) reads some specific characteristics by UUID.

lhwa561, 2017-07-12:
I have read through the example however the program freezes at 
```
string service_uuid("the uuid");
auto tinyb_service = sensor_tag->find(&service_uuid);

```

impala454, 2017-07-12:
And you verified you can connect to the device and read the same characteristic via gatttool?

lhwa561, 2017-07-12:
yes when i used gatttool and used `char-read-uuid` I could see that I was recieving hex values of the data i am sending from the tiny b

impala454, 2017-07-12:
Right but in the async example you used the full UUID for both the service and characteristic?

lhwa561, 2017-07-12:
The uuid i used the snippet was the uuid of the nordic uart service. The example in async uses a service uuid first then gets a characteristic uuid after which i was using too, however the program stops when i try to get the service using the service uuid.

```
std::string service_uuid("nordic service uuid");
std::cout << "Waiting for service " << service_uuid << " to be discovered" << std::endl;
tiny_service = sensor_tag->find(&service_uuid);
std::cout << "Service Discovered" << std::endl;
auto value_uuid = std::string("tx_uuid");
auto tx_value = tiny_service->find(&value_uuid);
```
So basically I get the cout at waiting for service but not the cout at service discovered and the program just stops after the first cout

impala454, 2017-07-12:
Ah, those nordic uart services can be difficult sometimes.  Have you tried making sure the device is cleared from any OS based Bluetooth managers?   Try the following to make sure they're removed 

    bt-device -r xx:xx:xx:xx:xx:xx

lhwa561, 2017-07-12:
Yes I made sure that the device is not connected to anything other than when the program is running

lhwa561, 2017-07-13:
Also to add, I am able to see the list of UUIDs through
```
auto uuid_list = sensor_tag->get_uuids();
if (!uuid_list.empty()) {
    cout << "UUID LIST: " << endl;
    for (int i = 0; i < uuid_list.size(); i++) {
        cout << uuid_list[i] << endl;
    }
    cout << "END OF LIST" << endl;
}
else {
    cout << "UUID_LIST is empty" << endl;
}
```
But 
```
auto service_list = sensor_tag->get_services();
if (!service_list.empty()) {
    cout << "Service List size: " << service_list.size() << endl;
}
else {
    cout << "Service List is empty" << endl;
}
```
gives me Service list is empty

which is making me very confused.

lhwa561, 2017-07-16:
I have solved this issue by getting a new pi 3. It seems to have been a device issue.

closed 2017-07-16 by lhwa561