intel-iot-devkit/tinyb #110: g_dbus_interface problem with BlueZ-5.46

crfrancis, 2017-08-07:
After my previous problems with read_value() (#106) I have set up a completely new Raspbian system, where, with a vanilla version of Raspbian Jessie (2017-07-05) I installed BlueZ-5.46 (and verified its correct with both bluetoothctl and bluetoothd showing this version number, and all other aspects running). Then, after installing cmake-3.8.2 I installed the current tinyb (hard to find the version number but it seems to be 0.5). 

Now, with the hellotinyb example I can find the SensorTag, but when it continues, trying to get the list of services, I get:
`(process:1621): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed`

I get an equivalent message with my own code, and at the same point where it is looking for services: 
`(process:1583): GLib-GIO-CRITICAL **: g_dbus_interface_get_info: assertion 'G_IS_DBUS_INTERFACE (interface_)' failed`

I don't think I've seen this behaviour before. Is it an issue with BlueZ-5.46? Or is it a dependency not up to date?

cheers,
Richard

petreeftime, 2017-08-08:
I think it might be an issue with libglib2.0-0 . What version do you have installed? Did you compile tinyb from sources? 

```dpkg -s libglib2.0-0``` will show you the version.

crfrancis, 2017-08-08:
I have version 2.42.1-1, and yes, I compiled tinyb from source. I checked the time of the downloaded sources and saw that it was the latest one (you had updated one of the examples a few hours before).

However, while I'm still having problems, this particular issue seems to have gone away (limited testing) since I applied the solution at the end of issue #43. Obviously I should have checked the issues before posting. 

Nevertheless, as I mentioned, I do still have problems: now it's read_value() timing out. write_value() and notifications continue to work as expected.

cheers,
Richard

closed 2017-08-12 by petreeftime