intel-iot-devkit/tinyb #115: Cannot close stdin, stdout stderr to make a daemon

westsuhanic, 2017-08-23:
Hello:

I have written a program using tinyb that reads all the sensors. Everything works as expected.
I am now trying to daemonize the program so that it can run when the machine boots.
Accordingly to get a program daemonized you must among other things close the open file descriptors
in the child via calls to close( 0 ), close( 1), close( 2).

However the issue I am having is that if I close all three the program stops working. If I only close 2 of the 3 the program works as expected. What am I missing? Is it a D-Bus issue? Any suggestions would be appreciated?

thank you,

west suhanic

ptzafrir, 2017-09-05:
When closing all the file descriptors Linux OS will kill the process. Have you made sure to call `setsid()`?

westsuhanic, 2017-09-05:
Hello:

Yes I called setsid. I have to leave one of stdin, stdout or stderr open.
Is the D-Bus code dependent on any of these?

thank you,

west suhanic

petreeftime, 2017-09-07:
DBus communicates over UNIX sockets, so there shouldn't be any requirements to keep any of the std* filedescriptors open, and there's plenty of DBus daemons.

Did you try running your process with GDB or strace and see if it has an error?

westsuhanic, 2017-09-07:
Hello:

I will run it with strace/GDB and get back to you.

thanks,

west suhanic

westsuhanic, 2017-09-19:
Hello All:

I discovered what was causing my daemon to exit without so much as a log entry anywhere. Me.
I was trying to close stderr while trying to write to stderr. This happened just before opening the
connection to syslogd.

My mistake. I apologize. All is well.

thank you,

west suhanic

closed 2017-09-19 by westsuhanic