intel-iot-devkit/tinyb #121: Segment fault after the very last commit

vkolotov, 2017-09-21:
On a clean VM (ubuntu) install I'm getting a segment fault. This is most likely due to the very last commit made by @ptzafrir: b5e293bb3c11abc21327f5c18d79436fab167038. See log file attached. 
[hs_err_pid8936.txt](https://github.com/intel-iot-devkit/tinyb/files/1322501/hs_err_pid8936.txt)

If I checkout the previous one @maggu2810 cb7ada6065a7226aa26dafd2c880124e115afc23, then it works just fine.

ptzafrir, 2017-09-22:
Could please provide more information as to what did you do prior to the segfault?
Or even better, could you provide a minimal code example that reproduce the issue?

ptzafrir, 2017-09-22:
@vkolotov Thank you for reporting this.
I found the problem and submitted PR #122

petreeftime, 2017-09-24:
Merged fix. Thanks @ptzafrir.

closed 2017-09-24 by petreeftime