intel-iot-devkit/tinyb #123: TinyB unable to find service

andy-huaan, 2017-09-26:
Hello everyone!
when I'm call `device.getServices().size() ` return 0，but when  I'm call `device.getUUIDs().length` return 4.
please tell me how to get `BluetoothGattService`  object
**Here's my code**
```
public static void main(String[] args) throws Exception {
		BluetoothManager manager = BluetoothManager.getBluetoothManager();
		manager.startDiscovery();
		List<BluetoothDevice> list = manager.getDevices();
		for(BluetoothDevice device : list){
			if("08:7C:BE:87:75:26".equals(device.getAddress())){
				System.out.println("name:"+device.getName()+
						" services:"+device.getServices().size()+
						" UUIDs:"+device.getUUIDs().length);
				
				break;
			}
		}
	}
```
system information：ubuntu16.0.4、bluez:5.9、BLE Device: [xicoo aTM07](http://www.xicoo.com.cn/?aTM)
I checked the [TROUBLESHOOTING.md](https://github.com/intel-iot-devkit/tinyb/blob/master/TROUBLESHOOTING.md) and all set.

On the other hand, I can get services using bluez `gatttool` 
(1) gatttool -b 08:7C:BE:87:75:26 -I
(2) connect
(3) primary
**Returns four entries**
attr handle: 0x0001, end grp handle: 0x000b uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x000c, end grp handle: 0x000f uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x0010, end grp handle: 0x0016 uuid: 0000fee8-0000-1000-8000-00805f9b34fb
attr handle: 0x0017, end grp handle: 0x001c uuid: 0000fffa-0000-1000-8000-00805f9b34fb

daviessm, 2017-10-23:
Are you calling this immediately after the device has been discovered? I found that the services are populated for some devices asynchronously after connecting and I needed to do this (Java 8):

      LocalDateTime startTime = LocalDateTime.now();
      while (LocalDateTime.now().isBefore(startTime.plusSeconds(30))) {
        if (this.device.getServicesResolved()) {
          logger.debug("Got characteristics after " + ChronoUnit.MILLIS.between(startTime, LocalDateTime.now()) + "ms");
          break;
        }
        try {
          Thread.sleep(100);
        } catch (InterruptedException ie) {
          throw new BluetoothException("Interrupted in Thread.sleep()", ie);
        }
      }
      if (!this.device.getServicesResolved()) {
        try {
          this.device.disconnect();
          this.device.remove();
        } catch (tinyb.BluetoothException bte) {
          throw new BluetoothException("Unable to disconnect from " + this.toString(), bte);
        }
        throw new BluetoothException("Unable to get services for " + this.toString());
      }

mqaisar, 2018-03-13:
I am working on rapberry p1 3 with SensorTag CC2650STK. I have same issue. I can connect with sensorTag but services are not discovered: When I use Please have a look over the code and suggest: 

import java.time.LocalDateTime;
import java.time.temporal.ChronoUnit;
import java.util.*;
import java.util.concurrent.locks.*;

import tinyb.BluetoothDevice;
import tinyb.BluetoothException;
import tinyb.BluetoothGattCharacteristic;
import tinyb.BluetoothGattService;
import tinyb.BluetoothManager;

import java.util.concurrent.TimeUnit;

public class HelloTinyB {
    private static final float SCALE_LSB = 0.03125f;
    static boolean running = true;

    static void printDevice(BluetoothDevice device) {
        System.out.print("Address = " + device.getAddress());
        System.out.print(" Name = " + device.getName());
        System.out.print(" Connected = " + device.getConnected());
        System.out.println();
    }

    static float convertCelsius(int raw) {
        return raw / 128f;
    }

    /*
     * After discovery is started, new devices will be detected. We can get a list of all devices through the manager's
     * getDevices method. We can the look through the list of devices to find the device with the MAC which we provided
     * as a parameter. We continue looking until we find it, or we try 15 times (1 minutes).
     */
    static BluetoothDevice getDevice(String address) throws InterruptedException {
        BluetoothManager manager = BluetoothManager.getBluetoothManager();
        BluetoothDevice sensor = null;
        for (int i = 0; (i < 15) && running; ++i) {
            List<BluetoothDevice> list = manager.getDevices();
            if (list == null)
                return null;

            for (BluetoothDevice device : list) {
                printDevice(device);
                /*
                 * Here we check if the address matches.
                 */
                if (device.getAddress().equals(address))
                    sensor = device;
            }

            if (sensor != null) {
                return sensor;
            }
            Thread.sleep(4000);
        }
        return null;
    }

    /*
     * Our device should expose a temperature service, which has a UUID we can find out from the data sheet. The service
     * description of the SensorTag can be found here:
     * http://processors.wiki.ti.com/images/a/a8/BLE_SensorTag_GATT_Server.pdf. The service we are looking for has the
     * short UUID AA00 which we insert into the TI Base UUID: f000XXXX-0451-4000-b000-000000000000
     */
    static BluetoothGattService getService(BluetoothDevice device, String UUID) throws InterruptedException {
        System.out.println("Services exposed by device:");
        BluetoothGattService tempService = null;
        List<BluetoothGattService> bluetoothServices = null;
[... 167 lines cut ...]
            /*
             * The temperature service returns the data in an encoded format which can be found in the wiki. Convert the
             * raw temperature format to celsius and print it. Conversion for object temperature depends on ambient
             * according to wiki, but assume result is good enough for our purposes without conversion.
             */
            int objectTempRaw = (tempRaw[0] & 0xff) | (tempRaw[1] << 8);
            int ambientTempRaw = (tempRaw[2] & 0xff) | (tempRaw[3] << 8);

            float objectTempCelsius = convertCelsius(objectTempRaw);
            float ambientTempCelsius = convertCelsius(ambientTempRaw);

            System.out.println(
                    String.format(" Temp: Object = %fC, Ambient = %fC", objectTempCelsius, ambientTempCelsius));

            lock.lock();
            try {
                cv.await(1, TimeUnit.SECONDS);
            } finally {
                lock.unlock();
            }
        }
        sensor.disconnect();

    }
}

andy-huaan, 2018-05-14:
@daviessm  Thanks，delay some time can solve

closed 2018-05-14 by andy-huaan

Jojiayur, 2018-06-28:
example for TI Sensortag (example -tinyb/examples/asynctinyb.cpp) to read temperature gets struck in below line
**Waiting for service f000aa00-0451-4000-b000-000000000000 to be discovered.**

None of the services gets discovered .Please help geeks.

mqaisar, 2018-06-28:
The issue was closed without solving my problem. I have boiled my blood but could not make it work for my, definitely there were errors in tinyB implementation.

mqaisar, 2018-06-28:
On Resbian there is a tool called "gatttool" installed. This is a command line tool and I had used Java process API to execute shell commands to access sensor.

joaopmarques66, 2019-04-07:
@mqaisar @EnricoHarder Have you guys found the solution to the problem? I Have the same problem. Gatttool recognizes my services but tinyb does not.

mqaisar, 2019-04-08:
Hi @joaopmarques66, I couldn't find the reason and couldn't come across any solution. I have written a small Java class which opens a Process to gatttool and then makes use of gatttool commands to read sensors. That's much easier than to use this heavy weight java lib which didn't work for me.

Jojiayur, 2019-04-08:
Just update to latest version of Bluez,it will work fine.

void-m4110c, 2021-04-25:
> Just update to latest version of Bluez,it will work fine.
> […](#)

Nope, it doesn't help at all and this is 2021 now...

AntistesSacrorum, 2021-04-28:
Hi
There is an API called **getServicesResolved**. 
If resolution were immediate there was no reason for that API. Since resolution may take some time one should implement the loop that waits for service resolution or use **enableServicesResolvedNotifications** callback.

Small code snippet that worked for me.
`
             System.out.println("allow 1 minute for service resolution");
            for (int i=0; i<60; i++) {
                System.out.println("attempt #" + i);
                if (device.getServicesResolved()) {
                    System.out.println("services resolved");
                    break;
                }

                try {
                    Thread.sleep(1000);
                } catch (InterruptedException e) {
                    e.printStackTrace();
                }
            }`

if-zimops-then-if-then-else, 2022-05-18:
> Hi There is an API called **getServicesResolved**. If resolution were immediate there was no reason for that API. Since resolution may take some time one should implement the loop that waits for service resolution or use **enableServicesResolvedNotifications** callback.
> 
> Small code snippet that worked for me: ``
> 
> ```
>         for (int i=1; i<60; i++) {
>             System.out.println("attempt " + i);
>             if (device.getServicesResolved()) {
>                 System.out.println("services resolved");
>                 break;
>             }
> 
>             try {
>                 Thread.sleep(1000);
>             } catch (InterruptedException e) {
>                 e.printStackTrace();
>             }
>         }
> ```
> 
> ``
> 
> Output: `attempt 1 attempt 2 attempt 3 services resolved`

I've tried that but somehow it never resolves and I'm stuck in the loop.